stuartpittaway/diyBMSv4ESP32 #156: possible to do just a single relay pulse rather then on/off every second?

gazzaman2k, 2022-12-30:
hi stuart, is it possible to have the relay just pulse once rather then on and off constantly i have a shunt that once powered trips the breaker and thats all it needs, as it currently stands i hear the shunt powering every second on and off like a indicator 

thanks and happy new year

stuartpittaway, 2023-01-01:
Hello, it shouldn't do this unless the rules are triggering on/off every second - can you adjust the hysteresis value to avoid this?

gazzaman2k, 2023-01-01:
i set my max battery temp to 5c as a test battery was at 12c so the rule was always triggered (green box) but when i set the relay type to pulse, it clicks the relay 1 on for a second then off, then on then off constantly every second sounds like a indicator in a car.

the rule was not being reset or turned off it was green throughout saying its triggered,

stuartpittaway, 2023-01-01:
Is this on the Esp32 or older 8266 version?

gazzaman2k, 2023-01-01:
the esp32 version

 Host name:[DIYBMS-006FCDE8](http://diybms-006fcde8/)

Processor: ESP32

Version: [4d240d0120df636fbe0bf136d1db2aa3e0a35694](https://github.com/stuartpittaway/diyBMSv4ESP32/commit/4d240d0120df636fbe0bf136d1db2aa3e0a35694)

Compiled: 2022-09-06T10:32:56.505Z

gazzaman2k, 2023-04-13:
Sorry yeah the bug is now fixed thanks :)

closed 2023-04-13 by gazzaman2k