stuartpittaway/diyBMSv4ESP32 #157: Controller reboots when touching the screen

deekoowee, 2022-12-30:
Enabling MQTT cause the system restart when touching the screen.
Disabling MQTT, no problem with the touchscreen.

stuartpittaway, 2022-12-31:
Hi, can you tell me a little more about the environment you are running? Controller and software version please?

deekoowee, 2023-01-01:
Hi, 
Source downloaded from git at 27.12.
Compiled and uploaded to controller (v4.2)
Version shows 'LocalCompile'
SDK Version: v4.4.2
My local MQTT server works fine and receives the MQTT packets from controller, when enabled at controller.
MQTT is used for HA
If MQTT enabled, I have to avoid touching the screen.

stuartpittaway, 2023-01-01:
Ok, can you capture the serial output (from the usb cable on the ESP32) ?

When it reboots it should spit out an error message.

stuartpittaway, 2023-01-01:
I've managed to re-produce similar behaviour to what you are seeing.

I've fixed this in the "pylontech" branch which will be released in the future. a1d92914849b18ada226945c3edc24c2d9d39b92

deekoowee, 2023-01-01:
Here's the serial capture, maybe you got it from your system too.
Capture is taken a little before reboot.

 (111704) diybms-webreq: API call: monitor2
W (112387) diybms-rules: Cell V>Max
I (112388) diybms-rules: Set warning 9
I (112612) diybms: Task 2
D (113608) diybms: Send addr=1, func=4, len=8
I (113746) diybms-mqtt: MQTT Payload for cell data
I (113771) diybms-mqtt: MQTT Payload for current data
I (113787) diybms-mqtt: General status payload
I (113801) diybms-mqtt: Bank(0) status payload
I (113809) diybms-mqtt: Rule status payload
I (113827) diybms-mqtt: Outputs status payload
D (113927) diybms: Recv 21 bytes, id=1, cmd=4
W (115389) diybms-rules: Cell V>Max
I (115390) diybms-rules: Set warning 9
I (115440) diybms-webreq: API call: monitor2
D (117608) diybms: Send addr=1, func=4, len=8
D (117912) diybms: Recv 21 bytes, id=1, cmd=4
D (118112) diybms: Task 3, s=0 e=1
I (118392) diybms-rules: Set warning 9
D (118753) diybms: total_free_byte=134900 total_allocated_byte=156100 largest_free_blk=94196 min_free_byte=120460 alloc_blk=539 free_blk=7 total_blk=546
I (118848) diybms-mqtt: MQTT Payload for cell data
I (118876) diybms-mqtt: MQTT Payload for current data
I (119180) diybms-webreq: API call: monitor2

assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)

Backtrace:0x40084121:0x3ffcbac00x4008e135:0x3ffcbae0 0x40094829:0x3ffcbb00 0x4008f0f5:0x3ffcbc30 0x400e6638:0x3ffcbc70 0x400d403e:0x3ffcbca0 0x400df1c0:0x3ffcbce0 0x4009175b:0x3ffcbd00

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[     7][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
I (52) diybms:

               _          __
  _|  o       |_)  |\/|  (_
 (_|  |  \/   |_)  |  |  __)
         /

CONTROLLER - ver:LocalCompile compiled 2022-12-30T18:09:44.137Z
ESP32 Chip model = 1, Rev 1, Cores=2, Features=50
I (64) diybms-hal: Configure I2C
I (67) diybms-hal: Scanning i2c bus
I (72) diybms-hal: Found i2c device at address 0x20
I (76) diybms-hal: Found i2c device at address 0x38
I (84) diybms-hal: TCA6416A not fitted, assume v4.2 board
I (86) diybms-hal: Found TCA9534A
I (87) diybms-hal: Found TCA6408
D (89) diybms-hal: Configure VSPI
I (94) diybms: ** Controller changed state from Unknown to PowerUp **
[   103][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (104) gpio: gpio_set_level(226): GPIO output gpio_num error
[   115][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (116) gpio: gpio_set_level(226): GPIO output gpio_num error
[   126][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
I (483) diybms-set: nvs_flash_init
I (485) diybms-hal: CAN driver installed.  Filter=1621098496 Mask=6291455
I (485) diybms-hal: CAN driver started
I (488) diybms: LittleFS mounted, total=589824, used=12288
I (492) diybms: Mounting SD card
I (494) diybms-hal: Mounting SD card
I (507) diybms-hal: SD card available

Press SPACE BAR to enter console WiFi configuration....

....................

No key press detected

deekoowee, 2023-01-02:
I confirm, this is working. No rebooting anymore, when touching the screen and MQTT is enabled.

closed 2023-01-02 by deekoowee