stuartpittaway/diyBMSv4ESP32 #158: Compilation fails during first week of the year

chron0, 2023-01-01:
**Describe the bug**

Compilation fails during first week of the year with:

```
include/EmbeddedFiles_Defines.h:22:49: error: expected primary-expression before ';' token
 static const uint8_t COMPILE_WEEK_NUMBER_BYTE = ;
```

which is probably due to buildscript_versioning.py line 91:

```
f.write(datetime.datetime.utcnow().strftime("%W").lstrip("0"))
```

when I run this in python console I get this result as well:

```
>>> import datetime
>>> datetime.datetime.utcnow().strftime("%W").lstrip("0")
''
>>> datetime.datetime.utcnow().strftime("%W")
'00'
```

chron0, 2023-01-01:
```
datetime.datetime.utcnow().isocalendar()[1]
```

could be an alternative here.

chron0, 2023-01-01:
```
f.write(str(datetime.datetime.utcnow().isocalendar()[1]))
```

works for me now.

stuartpittaway, 2023-01-01:
I fixed this earlier today, it's in the pylontech branch.

chron0, 2023-01-02:
ah perfect, I'll close this then :)

closed 2023-01-02 by chron0

N1c084, 2023-01-02:
Thanks for this .
When I compile a contrôler file and upload it by node mcu (1450ko) bin files it won't boot .
But when I compile and upload directly by platform io it work .🤔

Any idea ?

stuartpittaway, 2023-01-02:
The nodemcu files are complete flash images, not just the code. 

Platform IO only uploads the code, so the upload is smaller.

I use GitHub action to build the nodemcu files in the correct format.