stuartpittaway/diyBMSv4ESP32 #160: After update to "latest daily" can't configure the shunt

SergioRius, 2023-01-08:
**Describe the bug**
Today my controller died, so I replaced it with the spare one. The old was with a version from October 2021 and was working. I updated the new one with 2023-01-03 and also flashed all the modules to 4.40 @5k.
The modules operation work without problem. But I can't configure the shunt.
**Hardware/Software Versions**
Controller version (from PCB): 4.3
(below can be obtained from the "About" page in the controller web interface)
Processor: ESP32
Version: `e843bcb23f6aa0b0b75df78a05d59a0f5dc46bc7`
Compiled: 2023-01-03T10:07:59.024Z

**To Reproduce**
Steps to reproduce the behavior:
After activating the shunt feature, leaving the default settings but changing the shunt parameters to match my 300/50 and my battery, I hit Save, reports Ok, but after some seconds all the settings reset to blank.

stuartpittaway, 2023-01-09:
Can you update the code on the shunt to the latest version as well please?

SergioRius, 2023-01-09:
Yup! It will take a while because I have to take it out and it's "buried" in the system.
I don't know if I could update it in place using some trick, like disconnecting it from the actual shunt & the controller. ¿Would it be possible?

stuartpittaway, 2023-01-09:
As long as the current shunt is disconnected from the power, you can program the device in place using an updi programmer, set at5v

stuartpittaway, 2023-01-25:
There was a fault in the shunt code, it would work fine until it was first configured, then would stop responding.  This has been fixed today.

SergioRius, 2023-01-26:
> There was a fault in the shunt code, it would work fine until it was first configured, then would stop responding. This has been fixed today.

Nice! Thank you. I guess you're referring to the controller code? Or is it the shunt code?

I would also tell you that since I updated the controller loses connection to the modules at intervals. It may stay connected for a day or two, then fail for a day, and then repeat. The modules where also updated and running at 5k.
I had to disable the Victron integration because it thrown the system down every time.

On a side note/suggestion for improvement: is it posting an error/alert via canbus every time it starts balancing? Those appear repeatedly in the Victron system as alerts. I think you should not send those signals when balancing, but only when the matching rule activates.

stuartpittaway, 2023-01-26:
The shunt code changed, so I recommend updating to that version.

I'll disable the alert on the Victron CANBUS for balancing - will be in the next release.

SergioRius, 2023-02-10:
> 

Dear Stuart.
I haven't been able to test this because the controller doesn't sync with modules anymore, and I can't use it.
It has been dropping connection from time to time until now that doesn't detect them anymore.

stuartpittaway, 2023-02-11:
Did you change the connection speed in the controller to 5k (to match the modules) ?

SergioRius, 2023-02-12:
> 

Yes. It does match. The problem is intermittent. Yesterday it was working. Today is not.
I've checked all i could, wiring..

What the inter-packet gap should be for 14 modules? Is there any other config that can interfere with modules (and shunt) communications?

Now i have the shunt and canbus connectors unplugged.

SergioRius, 2023-02-12:
Now that I'm looking at it, it seems that it's receiving from module 1, isn't it?
[image]

stuartpittaway, 2023-02-12:
Ok, swap module 1 with another one and see if the problem moves. If it does, it's probably a faulty module

SergioRius, 2023-02-12:
I've already done that. When I flashed the new version on them, the first module refused to flash, so I retired it and picked a new one.
But I'll test to change it for another one.
Today lowered the interval gap to 2000 and left it alone. After some time it seems to had picked sync, now that I'm at the computer, I was looking at the web gui, with all working ok and it lost coms again in front of me.

SergioRius, 2023-02-12:
This is my voltage record for the module 0 for the past month. I can't see it's related with voltage.
[image]

stuartpittaway, 2023-02-13:
You mentioned it lost Comms in front of you, what message did you see?

SergioRius, 2023-02-13:
Just the red band that is saying something like "the controller is having trouble to communicate with modules"

Now it working again.

stuartpittaway, 2023-02-14:
Hows this looking now?

SergioRius, 2023-02-17:
> Hows this looking now?

At the time you asked, it was offline. I took the capture below. I waited but stills offline.
I would go to an earlier version, but I would be forced to also reflash all the modules.
I had two identical sets of controller boards and ESPs and my procedure was upgrading the spare one and then swap it with the installed. And that last became spare if all was working nice.

The fact is that initially upgraded but didn't change the speed settings nor reflashed the modules. And as the problems appeared, I went back to the old version (compiled 2021-10-05-10-49). That worked flawlessly.
But I thought it could be that I needed to flash the modules also to a newer version, so I tried again. And unfortunately in the meantime, the spare (good old) controller suffered an accident and got trashed. Esp and screen included. I couldn't get back.

So I'm pretty sure that there's some problem to the newer versions that make the boards incompatible with my ones that worked with 05/10/2021 time. And as first tried with 09/12/2022 it should be already present.
I asked on my country local DIYBMS fans club, but everyone confessed that they where not upgrading their devices.

Would you mind telling me what is the known good version/binaries just before the modules speed implementation?

[image]

stuartpittaway, 2023-02-17:
You definitely shouldn't be seeing 40,000 lost packets!!

I would have to dig through the GitHub history to go back that far.

As far as I can tell, the existing release along with updating the modules work fine, if in doubt use the slowest speed setting.

SergioRius, 2023-05-29:
It has passed some time and I'm still having issues. But I think I'll be better to do a little update on the progress.

This started as I  tried to update the firmware and tried the next step in the speed for the modules.
After the upgrade, I lost communication with my shunt and as I had the system integrated via canbus it stopped my victron system.
I became nervous after trying a lot of things without success and at the end, it seems that I managed to plug a comms cable in reverse despite the connectors tab and I got a good nuke. (it wasn't smoke, but a nice and sustained green flame coming from under the LCD.)
So I toasted my first controller. I replaced it with my spare one and also replaced the first module, that also seemed to be gone.
Unplugged shunt and deactivated it from the Victron configuration.
After that, It continued as the coms coming and going from the modules. The off interval increasing with time.
Seemed that the components in my modules weren't so happy with the increased speed and they started acting up. So I reprogrammed them with the standard speed but surprisingly the problems continued.

This last weekend I replaced the first module with a new one. That was the 3 change. I don't understand what is the problem, because if connected alone, it communicates well with the controller.

Took out the shunt to upgrade it's firmware as suggested, but as I flipped it, found burnt things on the underneath. The coms chip busted and a blown C5.

As I don't have another shunt to continue I left it aside. But anyways I'm trying to activate the canbus integration but I hadn't been able to make it work. I go to Charge/Discharge settings to activate it and in the cerbo to the services->BMS-Can port, and I can see that the Cerbo sends packets, but no reception. Also the DIYBMS shows sending packets errors.
Now I'm really, really scared to also had blown things on my Cerbo.

Does the canbus integration needs the shunt to have communication?
Is there any documentation to how to establish the CAN coms other than the video on the cable?
Thanks.

stuartpittaway, 2023-05-30:
Hi, you do really need to have working shunt to use CANBUS integration - as that provides the state of charge and current monitoring.

I've not heard of anybody having issues/burnt devices on the current shunt - did this get destroyed when you reversed the communication cables?

SergioRius, 2023-05-31:
> Hi, you do really need to have working shunt to use CANBUS integration - as that provides the state of charge and current monitoring.

Sorry if I'm insistent, but you mean that will not work well without a shunt or that I won't communicate at all?
As I'm trying to know if there's something else damaged I'm interested in testing the can communication (only the communication) but I don't have a shunt, now.

> I've not heard of anybody having issues/burnt devices on the current shunt - did this get destroyed when you reversed the communication cables?

Yep. I'll try to add pictures following:
As I connected RX to the controller, a 5cms flame came from below the screen pointing up. I thought it would eventually stop but it didn't. So I had to grab and pull the bottom cabling. It's a pity I didn't have a camera recording because it was truly spectacular.
The regulator turned into a drop of tin
[image]

Al other chipset was damaged
[image]
Of course the ESP didn't survive, nor the screen.

The shunt has the comms ADM chip damaged and the C5 condenser. I can't see something else damaged.
[image]

SergioRius, 2023-06-02:
Thank you very much Stuart.
As I could not establish CANBUS connection, I was concerned about the Cerbo been damaged when the surge happened. 
That was with my 2nd Controller (1st burned out), but yesterday I tried swapping the mainboard for another one that I got from friend, and that last one synced with the Victron system. So it seems that I have a bad canbus chip on my controller.

That leaves me with the recurring 1st module failing and not supporting any higher speeds. So I guess if I simply have a bad batch of chipset.

I'm currently awaiting for a new shunt and preparing a complete new battery. Soon I'll know if anything continues failing.

stuartpittaway, 2023-06-13:
@SergioRius I'll close this ticket for now until you are up and running again.

closed 2023-06-13 by stuartpittaway

SergioRius, 2023-06-19:
No disconnections since I replaced the 1st module.

I had a small fright when I accidentally enabled the DIYBMS as Controlling BMS instead of only Monitoring and a very sunny day came. The Victron was awaiting the command to stop charging the battery, but never came. So...

I think this issue can stay well closed.