stuartpittaway/diyBMSv4ESP32 #164: If you change the language, the rules are displayed incorrectly

moscito1010, 2023-02-03:
If you change the language, the rules are displayed incorrectly and the information "minutes since midnight" is missing.

stuartpittaway, 2023-02-03:
Which language, and which version of the code?

moscito1010, 2023-02-03:
oh sorry. the newest compiled firmware (master) and all languages without english.
in the pylontech brunch is it the same.

moscito1010, 2023-02-04:
Here a pic with pylontech branch
[image]

stuartpittaway, 2023-02-06:
Thank you for raising this problem.  I believe this is now fixed, in the pylon branch.

[image]

closed 2023-02-14 by stuartpittaway