stuartpittaway/diyBMSv4ESP32 #166: No Voltage1 and Range1 on Home Site with two Banks

moscito1010, 2023-02-14:
if i go to the HOME Site i can´t see the voltage and range from Bank1. I see only Bank0
in the Tiles site i see 2 banks

i use
latest release 2023-02-14

stuartpittaway, 2023-02-14:
Can you upload a screenshot?

moscito1010, 2023-02-14:
yes
[image]
[image]

stuartpittaway, 2023-02-14:
Ok, this is by design. When on the tiles page, click on the tile you want to be visible on the home page.

You can toggle them on/off

moscito1010, 2023-02-15:
oh, Stuart, I'm just embarrassed. I didn't know this function existed. Please excuse me.

stuartpittaway, 2023-02-15:
The feature was only added in this latest release.

closed 2023-02-15 by stuartpittaway