stuartpittaway/diyBMSv4ESP32 #167: no Minutes after midnight with Display settings in German

moscito1010, 2023-02-14:
i can´t see the minutes after midnight if i change the language to german

use latest release 2032-2-14

stuartpittaway, 2023-02-14:
Can you upload a screenshot please

moscito1010, 2023-02-14:
no problem
[image]
[image]

Bingo2023, 2023-02-14:
I just saw this...
check issue 168
https://github.com/stuartpittaway/diyBMSv4ESP32/issues/168

It is same route cause. Translated german rows are wrong.

stuartpittaway, 2023-02-15:
> Translated german rows are wrong.

The translation should be resolved/fixed now (see the screenshot in this issue) - the missing "minutes" value is still a problem though.

closed 2023-02-16 by stuartpittaway