stuartpittaway/diyBMSv4ESP32 #168: Rule-translation error

Bingo2023, 2023-02-14:
**Describe the bug**
error in german translation.
See screenshots of rule page for details.

**Hardware/Software Versions**
Controller version (from PCB): 4.5
(below can be obtained from the "About" page in the controller web interface)
Processor: ESP32
Version: v4.4.1-1-gb8050b365e
Compiled: Compiled_Firmware_2023-01-03-10-09

**To Reproduce**
Steps to reproduce the behavior:
1. switch to german -> reboot
2. configure rules (in german language) -> reboot to make sure rules are saved
3. switch to english language -> reboot
4. compare rule settings

[image]

[image]

the two screenshots show the values of rules are not changed, just the translation is switched.
In my case the BMS is connected to a real 16S Battery. As you can see the rule "Pack over voltage" is triggered because translated rows are mixed up. Note duplicate rows of Timer 1 and Timer 2

Fixing this should be easy via adaption translation files.

**BUT can we do something to prevent this in future? In my opinion it is a security risk as this can prevent the BMS from protecting against Over/unter voltage [MAIN reason for a BMS].**

stuartpittaway, 2023-02-14:
Can you update to the latest release please (from today)?

smurfix, 2023-02-15:
Don't forget the nonsense at the top.

smurfix, 2023-02-15:
Should be "Not-Aus", "interner Fehler im BMS", "maximaler Strom (A)".

stuartpittaway, 2023-02-15:
Ok, label text amended

[image]

stuartpittaway, 2023-02-21:
@Bingo2023 can you confirm this is now fixed?

closed 2023-04-13 by stuartpittaway