stuartpittaway/diyBMSv4ESP32 #172: Cell voltages not transmited to Venus OS

red0909, 2023-02-16:
the new controller firmware at pylontech setting does not report min and max cell voltage to the venus os
also no module temperatures

dvcc at the venus os does not work in pylontech mode. 
victron will force dvcc on(its a victron hardcoded rule for pylontech)

stuartpittaway, 2023-02-16:
Ok, are you using pylontech more on a victron system?

The pylontech CANBUS messages do not include as much information as the victron protocol.

If you are using victron kit, also use the victron mode.

red0909, 2023-02-16:
no i cant use victron mode because i have a mixed system sma and victron so pylontech should do it for both
maybe you can include the voltage readings in the pylontech protocol

but this does not explain why dvcc does not work on victron system in pylontech mode

stuartpittaway, 2023-02-17:
Ok - I don't have any documentation for the PylonTech CANBUS protocol which includes cell voltages.

DVCC shouldn't require those to function, just the charge voltage (which is supplied).

I wonder if the SEPLOS is mixing the CANBUS messages together (it shouldn't!) ?

red0909, 2023-02-18:
ok got the victron canbus working with my sunny island the prevent charging is inverted ive opened an other issue.

in victron canbus the cell voltages are working, so it is ok for me.

this issue can be closed if there is no way to add the cell voltages to the pylontech protocoll

stuartpittaway, 2023-02-20:
Do you have any better pylontech documentation?  

I'll take a look at #173 thanks for reporting

closed 2023-02-21 by red0909