stuartpittaway/diyBMSv4ESP32 #173: DVCC not working

red0909, 2023-02-17:
thats why my sma inverter crashes when i switch to victron canbus instead pylontech

the bms does not transmit cvl and ccl in victron mode

[image]

next problem
cvl ccl and dcl are correct at pylontech setting but the dvcc doesnt work too...

tried reboot of bms and venus os after settings change and tried to delete the bms from venus os but it still dont work

laserranger3000, 2023-02-18:
I have the same problem.
It works when I set the check box for "prevent charging", then the values are sent correctly to the gx device.
It seems that the setting is inverted.
-> I'm using 2023-02-16
-> I imported my old settings (maybe that has an influence?)

HerrFrodo1, 2023-02-18:
Yes, the Setting is inverted. I use 2023-02-16 too, without imported settings. 
If Prevent charge is aktivated, the cerbo gets the right values.

red0909, 2023-02-18:
ok i have activated "prevent charging" now the values are there but my charge controller (450/100) still wont charge in dvcc mode

not in pylontech and not in victron mode
i know it charge because the old style canbus in the old firmaware was working with dvcc charging

stuartpittaway, 2023-02-20:
This should be fixed now - you will need to switch "prevent charging" off.

red0909, 2023-02-20:
now with [Release-2023-02-20] at victron canbus the dvcc work

but at the pylontech canbus, dvcc still doesnt work
if battery id B009 is used then venus os will force dvcc on(it can be manualy edited in some venus os fils not to force it on)

stuartpittaway, 2023-02-21:
> but at the pylontech canbus, dvcc still doesnt work

I've never used PylonTech battery on Victron kit - I would normally advise not to and use the genuine Victron protocols instead.

Are you using the DIYBMS current monitor?  If not, perhaps it requires those missing messages (current, state of charge etc.)

red0909, 2023-02-21:
i have the shunt installed and running with latest firmware

but pylontech on victron should support dvcc, it is supported by victron.
maybe you can fix this in future, for me now the victron canbus does the job.

closed 2023-04-13 by stuartpittaway