stuartpittaway/diyBMSv4ESP32 #175: Victron VE.CAN - CVL handling

Bingo2023, 2023-02-20:
SW-Version: [Release-2023-02-20-09-53](https://github.com/stuartpittaway/diyBMSv4ESP32/releases/tag/Tag-2023-02-20-09-53)

In general, on Victron (VE.CAN), setting "CVL" (Charge Voltage Limit) = 0 V means we have Batterie-Overvoltage -> discharge battery with full power.

- So I we set option "Prevent charging" -> CVL goes to 0V and battery gets discharged rapidly. I tried to avoid damage by setting CCL = 10A but also this gets ignored (in my case discharging w/ 100A.... :( )

- I guess same for Option "Prevent discharging" but I did not feel like trying this.... 

[image]
[image]

Bingo2023, 2023-02-20:
Another thing regarding CVL handling:

I experienced this today afternoon while charging battery.
CVL gets calculated by some function (voltage-level changes during charging).
So it was not possible the charge battery to full SOC, I stuck at ~50% because charge-voltage is limited -> power will get reduced.

In my understanding (I work in automotive e-mobility area), we should simply send the voltage defined on this page via CAN (in the pictures 58,0 V).
All limitations is only done via current limitation.

To me it is not clear how the setting block "Discharging" is supposed to work.
I think we need something like this (green color). Limiting current toward lower end until 0A (usually linear dearating is enough).

[image]
[image]

stuartpittaway, 2023-02-20:
Have you updated to today's release? #173 is related to this

stuartpittaway, 2023-02-20:
Ok, I was unaware of the cvl zero issue, is this documented somewhere?

Victron recommend using voltage as the main limiting factor when charging battery, not current.

stuartpittaway, 2023-02-20:
The value of 3500mV for the linear curve is also too high, it should be around 3200

Bingo2023, 2023-02-20:
> Ok, I was unaware of the cvl zero issue, is this documented somewhere?
> 
> Victron recommend using voltage as the main limiting factor when charging battery, not current.

The control loop always has an voltage limit as setpoint that is correct, but there is also a second control loop that limits the current (value should come from limiting device can be charger or battery or whatever).
During charging the control loop is always limited in current until the very last few percent before 100% SOC.

Bingo2023, 2023-02-20:
> The value of 3500mV for the linear curve is also too high, it should be around 3200
Thx, I will try with 3200mV. I've also watched the new video and found the code provided by Matthias.
I will test voltage limitation more deeply in the next days. On my setup it VCL is less that 1V over battery voltage, sometimes even same level... this will almost prevent charging...

stuartpittaway, 2023-02-21:
> On my setup it VCL is less that 1V over battery voltage, sometimes even same level... this will almost prevent charging...

Ok, try lowering the "sensitivity" value then - however as long as there is a voltage difference, charging shouldn't be affected.

I'll modify the code to avoid using "zero volt" when charging is disabled/not wanted.

stuartpittaway, 2023-02-21:
@Bingo2023 I've updated the code with fixes for sending a zero voltage to Victron.

I've just noticed that the screenshot you uploaded had sensitivity set to "50" - that should be like "1.5" - probably the reason why you see such low charging voltages.

stuartpittaway, 2023-02-22:
> To me it is not clear how the setting block "Discharging" is supposed to work.

Presently, there is no dynamic control on the discharge parameters - discharge is simply prevented when the cell and/or bank voltage goes below the setting.

Bingo2023, 2023-02-22:
Hello Stuart,
thx for the extremely fast reaction :)
The release
[Release-2023-02-21-08-55](https://github.com/stuartpittaway/diyBMSv4ESP32/releases/tag/Tag-2023-02-21-08-55)
is working with Victron.
Also it seems that the proposed setting helps the chargevoltage. I'll test these days more and give feedback.

closed 2023-04-13 by stuartpittaway