stuartpittaway/diyBMSv4ESP32 #177: Manually setting SoC does not work

laserranger3000, 2023-02-22:
I tried setting the SoC manually to 100% (from 105%), but it didn't do anything.
Now it seems to be permanently stuck at 105%.
I tried it another time with 99%, but same behavior.

stuartpittaway, 2023-02-22:
Have you updated the firmware on the current shunt recently?  This only works on newer code versions - this is the latest release https://github.com/stuartpittaway/diyBMS-CurrentShunt/releases/tag/Tag-2023-01-25-12-26

laserranger3000, 2023-02-22:
I wasn't aware of the new shunt firmware, I guess that explains my issue :)

Did I break something with my mismatching firmware combination? As I said, the SoC does not seem to update anymore and is stuck at 105%

stuartpittaway, 2023-02-22:
I've completely re-wrote the SoC calculation in the new firmware - so give it a go.

stuartpittaway, 2023-03-10:
Hi @laserranger3000 did you get this working?

closed 2023-03-20 by stuartpittaway