stuartpittaway/diyBMSv4ESP32 #178: Platform IO compile error if using ZIP from GITHUB not git clone

stuartpittaway, 2023-02-22:
**Describe the bug**
As per this description....
https://community.openenergymonitor.org/t/current-monitor-shunt/17973/245?u=stuart

Compile using platformio after downloading ZIP version - looks like issue with build script python code(s)

virtuvas, 2023-03-20:
morning Stuart,
is this bug still on? I decided to update all my s/w and check what's new, so zip downloaded (as I always did) the code from github, updated platformio, loaded, compiled, uploaded to the second board I have as backup here at home without issues. Now if only I remember how to configure wifi...

virtuvas, 2023-03-20:
hm, maybe not... I get this:
```
CONTROLLER - ver:LocalCompile compiled 2023-03-20T09:02:47.930Z
ESP32 Chip model = 1, Rev 1, Cores=2, Features=50
I (50) diybms-hal: Configure I2C
I (52) diybms-hal: Scanning i2c bus
I (58) diybms-hal: Found i2c device at address 0x20
I (61) diybms-hal: Found i2c device at address 0x38
I (70) diybms-hal: TCA6416A not fitted, assume v4.2 board
I (71) diybms-hal: Found TCA9534A
I (73) diybms-hal: Found TCA6408
D (75) diybms-hal: Configure VSPI
I (79) diybms: ** Controller changed state from Unknown to PowerUp **
I (440) diybms-set: nvs_flash_init
I (442) diybms-hal: CAN driver installed.  Filter=1621098496 Mask=6291455
I (442) diybms-hal: CAN driver started
./components/esp_littlefs/src/littlefs/lfs.c:4191:error: Invalid block count (144 != 128)
E (451) esp_littlefs: mount failed,  (-22)
E (455) esp_littlefs: Failed to initialize LittleFS
[   465][E][LittleFS.cpp:95] begin(): Mounting LittleFS failed! Error: -1
E (466) diybms: LittleFS mount failed, did you upload file system image?
E (472) diybms: SYSTEM HALTED
```

stuartpittaway, 2023-03-20:
Use the upload filesystem option in platformio.

Alternatively, just use the precompiled version from github

virtuvas, 2023-03-20:
doh, had forgotten the upload filesystem completely! will do tonight (and right it down in my notes!)
Means that github download as zip works OK.
cannot use precompiled version as I'm adding some influxDB bits which I've not given to you for evaluation and inclusion, have to!

cheers

V.

stuartpittaway, 2023-04-13:
Issue #200 also relates to this.

Closing as fixed

closed 2023-04-13 by stuartpittaway