stuartpittaway/diyBMSv4ESP32 #181: Suggestion - Dynamic Balance Voltage

red0909, 2023-03-02:
i think it would be better to have a start balance voltage, deviation start and deviation stop option.
if one cell exceed the start balance voltage and the deviation it get balanced untill the deviation stop value.

example, my charge voltage is 3,5V  start balance will lbe 3,4V and start deviation 30mV, stop deviation 10mV.
this is common in other bms like seplos or jk, it works really well.

HerrFrodo1, 2023-03-02:
Good idea.  I also charge up to 3.5V per cell but still use an active balancer that shovels the energy back and forth.  That's probably why I only have a larger cell drift from 3.5V.  I no longer use the top balancing.  Without the active balancer, my ESS wouldn't work well either.  200A discharge, the next moment charge with 200A (of course only on sunny days).  Up to just before 3.5V per cell up to 180A.  56.1V end-of-charge voltage then ensures that the cells draw the current according to Ohm's law.  The passive balancer from the modules would only start at 3.55V, but that is never reached.

[image]

stuartpittaway, 2023-06-23:
I've added something similar to the all-in-one module (prototype in my last YouTube video) this isn't as easy for the existing single cell modules as they operate independently of each other.