stuartpittaway/diyBMSv4ESP32 #182: Pylontec over RS485

elmozwo, 2023-03-02:
Possibility to connect a Pylontech batterie via RS485 for inverters or Logger wich don't have a Can bus interface e.g. FSP Hybrid Power Manager and Plexlog.

**Protokoll description:**
[https://www.photovoltaikforum.com/core/attachment/77134-pylon-lfp-battery-communication-protocol-rs485-v2-8-20161216-pdf/](url)

If there are more informations required, just ask me. Possibly I can find them.

ruza87, 2023-03-31:
Hi Stuart, I have a simple implementation of Pylontech protocol over RS485 integrated into diyBMS code and running with the PIP5048 / Voltronic VM III inverter.

The integration into diyBMS isn't perfect, I didn't know how to make it work together with existing RX/TX queues in main.cpp, so I commented those lines (I'm not using RS485 current shunt) and it's calling uart_read_bytes / uart_write_bytes directly.

What's implemented: 
- reporting of charge voltage and current limits to the inverter
- alarm flags to stop charging or discharging.

What's not: 
- voltage and temperature monitoring per each cell
- capacity & SOC reporting

Let me know what you need, I can share the code.

ruza87, 2023-04-03:
Ok, will do.

stuartpittaway, 2023-04-13:
@elmozwo do you have anyway to test the changes @ruza87 has made?

I don't have the necessary inverters/equipment to test the RS485 protocol.

stuartpittaway, 2023-04-13:
@ruza87 can you help with some build instructions?

ruza87, 2023-04-14:
@elmozwo  Are you familiar with compiling of the sources in Visual Studio Code using PlatformIO addon? Clone the sources from https://github.com/ruza87/diyBMSv4ESP32/tree/pylon_rs485 and the next steps to build & flash are the same as in the original Stuart's repo.

Or you might try to use already compiled code from my (first, highly experimental:-) release. You can download it here https://github.com/ruza87/diyBMSv4ESP32/releases/download/v1.0.0/release_2023-04-14-09-07.zip and follow the original instructions to flash it: https://github.com/stuartpittaway/diyBMSv4ESP32/blob/master/ProgrammingHardware.md

Jestergnet, 2023-04-18:
How is it going? Also interested in this area...

stuartpittaway, 2023-04-19:
For people who want to test this (@Jestergnet and @elmozwo ) GITHUB actions have pre-compiled the release.

https://github.com/stuartpittaway/diyBMSv4ESP32/actions/runs/4740188233?pr=195

Look at the bottom of that page for "[DIYBMS-Compiled]"(https://github.com/stuartpittaway/diyBMSv4ESP32/suites/12332390974/artifacts/654416283)

And use this file for the controller.
[image]

Jestergnet, 2023-04-19:
Has anyone already tested the code? After the RS485 firmware, I don’t show any signs at all. Diybms acting as a host? Who is contacting whom? Inverter to BMS or BMS to inverter?

Jestergnet, 2023-04-19:
> @Jestergnet I did. Mine is working with PIP 5048 (same as Voltronic VM III), the battery icon on the inverter is flashing, which means the communication is running OK. The inverter is a host, asking BMS for a reply.
> 
> Did you select _PylonTech RS485_ emulation on the diyBMS "Charge/Discharge configuration" page? Also, is your RS485 speed set correctly? For this inverter it needs to be set to 9600 baud, you can do it on "Current &Voltage Monitoring" page.
> 
> Regarding the screen issue - I don't have the screen attached so I cannot tell.

Ok, I'll try.

elmozwo, 2023-04-24:
Finaly I also get some time to test. It looks good! The only thing is that it produces many warning popups about discharging/ charging is prevented. Even check or uncheck the boxes will not change anything.

elmozwo, 2023-04-25:
I just have the two sensors from the batterie modules. Do I need a additional one?

ruza87, 2023-04-28:
Two sensor are fine. Can you check all the rules to see if any of them is preventing the charging? Is the communication between your inverter and diyBMS working? What inverter model do you have?

Jestergnet, 2023-06-14:
Found a problem, after activating PylonTech RS485, the screen loses its refresh rate. If you disable this option, the screen returns to the normal refresh rate of the image. In order for one page to load, it probably takes 20 seconds, or even more ... 
#[image]

stuartpittaway, 2023-06-23:
From a quick review of the code @ruza87 I think you are in a near-infinite loop waiting for serial data to arrive.

The original code uses this ```ulTaskNotifyTake(pdTRUE, portMAX_DELAY);``` as a delay (part of the RTOS) until the transmit function triggers (notifies) it.

ruza87, 2023-06-27:
Oh, I see it now, thanks @stuartpittaway. I call `uart_read_bytes` when the RX buffer contains at least pylon protocol header. This is tested in `uart_get_buffered_data_len`. But, if the buffer is empty (or contains less data) the task is not suspended which leads to quick looping... I'll add `vTaskDelay` to pause the task in that case.

Jestergnet, 2023-06-27:
> Oh, I see it now, thanks @stuartpittaway. I call `uart_read_bytes` when the RX buffer contains at least pylon protocol header. This is tested in `uart_get_buffered_data_len`. But, if the buffer is [...]

Let me know when you make a fix, I'll check.

ruza87, 2023-06-30:
I fixed it in my repository. You can download the new release from here (https://github.com/ruza87/diyBMSv4ESP32/releases/tag/v1.0.1) or preferably wait for @stuartpittaway to approve GITHUB actions to create automated release.

Jestergnet, 2023-06-30:
> I fixed it in my repository. You can download the new release from here (https://github.com/ruza87/diyBMSv4ESP32/releases/tag/v1.0.1) or preferably wait for @stuartpittaway to approve GITHUB actions to create automated release.

Hi, thanks for the fix. Checked, now the screen does not slow down.