stuartpittaway/diyBMSv4ESP32 #185: Current & Voltage Monitoring - Basic Settings, saving mystery

deekoowee, 2023-03-15:
**Describe the bug**
When saving the basic settings, settings will be changed. Tried with several computers and browsers. Always the same result.
I'm not sure, but few weeks ago settings have saved correctly (battery was not fully charged). Now the battery is fully charged to top limit 58.8V and saving problem started.

**Hardware/Software Versions**
Controller version (from PCB): 4.2
Processor: ESP32
Version: a21d13edc38a212d7bd5366c9cbb3858a885eaa1 
Compiled: 2023-03-01T09:16:31.399Z

**To Reproduce**
Steps to reproduce the behavior:
1. Set the values
[image]
2. Click on 'Save basic settings'
3. Result
[image]

deekoowee, 2023-03-16:
Tested with earlier version 14th Feb 2023 and worked ok.
It doesn't change the values.

stuartpittaway, 2023-03-16:
Thanks for reporting this bug, think I know what it is.

stuartpittaway, 2023-03-16:
Hi @deekoowee 
I've just merged into the master branch some changes to fix this issue.  Would you be able to confirm it now works using that version?

https://github.com/stuartpittaway/diyBMSv4ESP32/releases/tag/Tag-2023-03-16-15-47

deekoowee, 2023-03-16:
Hi @stuartpittaway 
I programed the new version into controller.
Now the saving works ok and do not change the values.
Thanks for this fix.

stuartpittaway, 2023-03-16:
Perfect, thanks for the feedback

closed 2023-03-16 by stuartpittaway