stuartpittaway/diyBMSv4ESP32 #188: Controller starts new

moscito1010, 2023-03-20:
Hello Stuart
Somehow the latest firmware has a problem. 
The SD card cannot be mounted, the communication keeps breaking off. If I want to flash firmware on a module with the controller, the controller crashes. In between, the controller restarts automatically, this also happens when you press the buttons under the display. I haven't changed anything on my system.
The controller is from you from the Patreon campaign, the modules are v4.40 in the 1.9F 123mm design for LFP cells.
Anyone else have these problems?

stuartpittaway, 2023-03-23:
Hi @moscito1010 

The controller shouldn't be rebooting - in fact the latest release is the most stable - I'm over 50 days uptime and counting.

Can you try programming the modules with the TFT display REMOVED ?

A few people have suggested that the TFT screen seems to cause problems for them.

moscito1010, 2023-03-23:
Oh damn, I didn't even think of that. The display can generate voltage dips. I will try that immediately. It might make sense to build an adapter that allows me to power the display externally. Stuart, thank you very much for this information

stuartpittaway, 2023-03-24:
If you are powering the controller using the +5V/2Amp connection on the PCB, this shouldn't be a problem.

Did removing the LCD fix things?

moscito1010, 2023-03-27:
Sorry for the late reply, I've been very busy.

So without a display, each of the 32 modules (v4.40 in 1.9f LFP design) could be written with the firmware and the modules are also recognized again.
I flashed the standard firmware, not the 9600 baud or something. I'll test that later.
After that I was able to use the display again. The power supply delivers clean 5V with a maximum of 10A. So that should be enough as a power source.

stuartpittaway, 2023-03-28:
Thank you for testing - could be a potential issue with the controller circuit board (its design).

Can you tell me what length cable you used (6 pin ISP cable) and also did you remove all the cables to the modules before flashing?

moscito1010, 2023-03-28:
150cm cable
like this
https://www.amazon.de/Flachbandkabel-Pfostenbuchsen-6-polig-Atmel-Programmer/dp/B019IK8EOQ

stuartpittaway, 2023-03-28:
150cm cable is very long!!  I normally only use a 10cm cable.

stuartpittaway, 2023-03-29:
There is a limit of 128 modules in the system, so technically 8 banks are possible - although I don't think anyone has done this yet!

maciejkrasuski, 2023-04-11:
> controller restarts automatically, this also happens when you press the buttons under the display. I haven't changed anything on my system.
> The controller is from you from the Patreon campaign, the modules are v4.40 in the 1.9F 123mm design for LFP cells.
> Anyone else have these problems?

I have similar problem. Just touched the screen to turn it on and then controller reboted (I don't have these buttons on the controler board). Also after last update I cannot mount sd card. I use the same 1.9F boards as a cell monitors.

virtuvas, 2023-04-11:
had similar problems with the screen, turned out the diode was soldered slightly high and was fouling (shorting) on the sdcard, draining the powersupply and rebooting the board.
explained in #125 
temp solution was to use a piece of insulation tape 😁

maciejkrasuski, 2023-04-11:
> 

Thanks for your reply but in my case there is nothing that could couse short on the back of the screen. And also it doesn't do it every time i touch it ;(

moscito1010, 2023-04-11:
@Maciejkrasuski So you're also using the 1.9f modules.
I will equip the battery packs with the old 4.21 modules in the next few days.
I now want to know whether the modules are possibly the reason for these faults.
Because I can say one thing with certainty, the controller ran without any problems with the 4.21 modules. Now I have O.O.S errors, constant communication errors and reboots. The system has become so unreliable that I had to turn it off.
With 16 modules it was still within limits but with 32 modules it is unreasonable.

And believe me, I've tried everything that's possible. Flashed firmware with all 3 possible speeds, used even better shielded cables. I checked the modules for errors with the lupa to see if incorrect components from JLCPCB were used. All 1.9f modules were exchanged by me.
None of that was helpful. The last option that occurs to me is to exchange the modules for version 4.21 because I have already exchanged the controller and the smart shunt.
Possibly the 1.9f modules are also incorrect constructions. The things have not been tested by Stuart and it is not impossible that the things have a design flaw.

If so, I would be willing to donate enough money to allow Stuart to create modules in the LFP design based on version 4.40, which he then approves.

maciejkrasuski, 2023-04-11:
> 

I don't think that 1.9f modules are the reason by itself. I will bet on the number of modules. When I had only sixteen cells connected to the controler almost everything worked well. But right now I also get a lot of oos and crc errors. But the strange part is that after todays crash the error count is still 0 - after more than 8 hours.
Right now I only have bad packet count on last three modules - I even switched one to new and there are still bad packets.

Nevertheless I think that you should start with newest firmware and only 16 cells and see if it helps before you will change the cell monitors.

moscito1010, 2023-04-12:
I always have the latest firmware on the modules and the controller.
There were no problems with the 4.21 modules, I switched to the 1.9f because it looks very tidy and I find the modules practical.
The fact that the last 3 modules always display many errors is also strange. If I change Bank0 to Bank1, the last 3 modules also have many errors. That's not normal.
After the many tests I've done, I'm always 90% sure that the 1.9f modules are the source of the problems.
I will have to swap all modules to verify that.

stuartpittaway, 2023-04-12:
Communications has always been a problem with the module design -  I do plan to fix this, but it would mean existing hardware won't work.

The best solution currently, is to use the slowest baud rate (2400) and twisted/shielded cables.  Also look at the modules page to see if one module has higher errors than the others.   

You can also put the controller into the "middle" of the comms chain - it doesn't have to be at one end - although the module addressing will then be different to what you have now, but this can help if you have 32 cells - put it in the middle.

moscito1010, 2023-04-12:
I do not understand what you mean. How can I operate the controller in the middle? Can I connect 2 cables to RX/TX?
On the other hand, I don't want to use any modules that generally have communication problems. So it is clear that I will switch completely to the 4.21 modules. I will desolder the ATtiny's from the 1.8f modules. If the 1.9f are generally problematic, that's core scrap for me and will be disposed of. They've annoyed me long enough :-)

If you say you want to fix the problem, does it stay with the current attiny's? I still have 100 of those lying around, that would be quite annoying if they were no longer useful.

moscito1010, 2023-04-12:
Ok, Stuart, I don't get it. Please make me a sketch.

I never had a problem getting the attiny. jlcpcb never has any, but I always got sufficient quantities from mauser or digikey.

Using twisted cables is a matter of course. My hobby is HAM and RC. I have meters of suitable cable :-) Disturbance on "long-distance" flights due to unsuitable cables would be a disaster.