stuartpittaway/diyBMSv4ESP32 #189: PylonTech emulation - battery quantity

deekoowee, 2023-03-22:
Hi,
Setting up my new inverter and propably I found battery bank calc error.
Seems, that code do not make correct calculation at line 178 and firmware defaults battery banks to 1 at line 183 ?

**Hardware/Software Versions**
Controller version (from PCB): 4.2
Processor: ESP32
Version:  e861a254201a9526f29e00371e471e47b71b5a06 
Compiled: 2023-03-16T15:45:16.346Z

**To Reproduce**
Steps to reproduce the behavior:
1. Selected PylonTech emulation with 830Ah capacity at Charge/Discharge configuration.
2. Inverter site, selected model 3000C as described in code.
3. Inverter informs only 1 battery bank connected.
Changing the battery capacity to 148Ah or 296Ah did not change battery bank amount.

stuartpittaway, 2023-03-29:
This is now fixed in release https://github.com/stuartpittaway/diyBMSv4ESP32/releases/tag/Tag-2023-03-29-08-24

deekoowee, 2023-03-30:
I upgraded the controller firmware to 4e72db94af375c391bad47b9b98507b3271b069d

1) upgrade defaults the 'Nominal Battery Capacity' to 280
2) changed back to 830 and inverter 'see' 11 batteries
3) tested the charging and max current is is inverter's top 120A

I would say, that this fix works.

stuartpittaway, 2023-03-30:
Thank you for testing

closed 2023-03-30 by stuartpittaway