stuartpittaway/diyBMSv4ESP32 #191: problems with communication if i use "storage & logging"

moscito1010, 2023-03-27:
**Describe the bug**
A clear and concise description of what the bug is.

Controller from the Patreon Deal
Version: 17.march.2023
Precomp Image

Is it possible that the SD card is not recognized in the current controller firmware? There are also problems with communication when I want to use the "storage & logging" option

[image]

stuartpittaway, 2023-03-28:
Hi, thanks for the detailed logs.

ESP32 should support up to 4GB of capacity - anything larger than that should still work, it just won't be "seen".

Have you tried simply re-formatting the card again (FAT32 format) ?

The SD card library/code in the controller gets updated automatically with the "core" of the ESP32/Arduino so I don't specifically change anything - although this does look like a potential bug.

moscito1010, 2023-04-01:
I just can't get around to it, I'll try on Sunday.

moscito1010, 2023-04-03:
well, but now....
Today I had some time and let the Raspi write it down.
putty1.log is without SD card
putty2.log is with inserted SD card

The following fact is striking.
I supply the EPS32 controller with 5v from a switching power supply via the terminal block and not via USB.
To log I have to plug a USB cable into the USB port of the ESP32.
As soon as I operate the controller without an SD card AND use an additional power source (the USB cable), does the controller run more stably?!?
With the SD card inserted, the controller is unusable and keeps restarting or freezing.
[putty1.log](https://github.com/stuartpittaway/diyBMSv4ESP32/files/11139058/putty1.log)
[putty2.log](https://github.com/stuartpittaway/diyBMSv4ESP32/files/11139059/putty2.log)

stuartpittaway, 2023-04-03:
In the first log, there is a message Found TCA6416A, which is correct.

However, in the second log, you have hundreds of errors in the log files like this....

HAL_ESP32::WriteTCA9534APWROutputState()

This indicates the 6416A chip was not found when the SD card was inserted.

Can you upload a photo of the controller, particularly the version number.

moscito1010, 2023-04-03:
This is the controller from the Patreon campaign. You sent it to me in the mail. Purple PCB

stuartpittaway, 2023-04-04:
Ok, is this the one with the extra wire soldered onto the reverse?

moscito1010, 2023-04-04:
nein

stuartpittaway, 2023-04-12:
Reviewing the second log file, the message ```***ERROR*** A stack overflow in task sdlog has been detected.``` has been reported.

stuartpittaway, 2023-04-13:
The sdlog overflow error should have already been fixed by commit 21d4879ab7d5aae70540305a8ea67d19289c856e

This is part of the next release.

stuartpittaway, 2023-04-17:
The release was pushed up last week - can you test and report back please?

moscito1010, 2023-04-26:
Sorry my response is so slow but I have so much work to do.
Today I had a few minutes and brought one of the two battery storage systems to my workshop.
I flashed the new firmware to the ESP32 and screwed the battery pack back on with 16x 1.9f modules
The controller still crashes as soon as I tap the screen, that's all I can tell. The number of O.O.S errors has dropped to ZERO. But there are only 16 and not 32 modules.

Tonight I will prepare Zen Buddhist 16 pieces 4.21 modules to use in the test. Just to confirm the 1.9f modules as a possible bug. That's only possible with a direct comparison.

stuartpittaway, 2023-04-27:
If you don't need it, can I suggest removing the TFT display? I've got 3 running at home and none of those crash on touch, could be a faulty esp32, a soldering issue or a faulty TFT

stuartpittaway, 2023-06-23:
How's the system operating @moscito1010  ?

moscito1010, 2023-06-24:
Not Stuart at the moment. I am using a Battrium BMS until I get the 4.50 modules from JLCPCB. I destroyed or gave away all the 1.9f modules and I didn't have enough of the other modules. The 4.50 modules will arrive next week. Unfortunately no attinys were available, I got them from another source. When I have the boards I bring the ICś and the boards to a service provider and have the attiny's soldered on. They're too small for me, I can't solder them without a camera and magnification. I will probably have to buy helpful technology if I want to solder it myself in the future :-)

stuartpittaway, 2023-10-30:
I'm closing this issue for now as I think the SD card logging function is working fine in the latest release.  Please raise a new issue if this is incorrect.

closed 2023-10-30 by stuartpittaway