stuartpittaway/diyBMSv4ESP32 #192: Coding help for Freedom Lite

deekoowee, 2023-03-28:
Hi,
My inverter supports [Freedom Lite](https://www.freedomwon.co.za/) storage.
It seems to use similar protocol as PylonTech have.
Dynamic charging is supported and I can see the charging current changes at inverter's hidden webpage.
I have tried PylonTech 3000 setting at inverter configuration, but it have some mystery charging issues.
That's why, I go to Freedom Lite. Now I can charge much more quicker (120A).

I don't have enough programming skills to make modifications to [pylon_canbus.cpp](https://github.com/stuartpittaway/diyBMSv4ESP32/blob/master/ESPController/src/pylon_canbus.cpp) file.

Following important changes saves Li-Ion battery life;
1) charging voltage; Maybe it could be same as 'Battery charge voltage' in Charge/Discharge configuration page. I'm not sure, if it is line 37 in code and is defaulted to 1. (Inverter defaults to 55,8V)
2) discharge voltage; Might be same as 'Battery discharge voltage minimum' in Charge/Discharge configuration page (line 40 ?), but changes doesn't do any difference at inverter's hidden page. (Inverter defaults to 46V)

What kind of changes I should do ?

stuartpittaway, 2023-03-29:
Hi @deekoowee, do you have documentation on the protocols Freedom battery requires?

If not, we can't make any code changes as we don't know what "language" the inverter expects to communicate under.

Its possible that your inverter is limiting the current on the PylonTech battery due to the other issue you mention #189

stuartpittaway, 2023-03-30:
I would recommend using the pylontech more on the inverter, as we know that's fully supported, I can't verify if the lite protocol does anything strange, and we don't want to risk the cells.

The state of charge needs to reset on fully charged battery voltage, take a look at the shunt settings on the controller

deekoowee, 2023-03-31:
Hi @stuartpittaway,
Some mystery things has happened again.
After yesterday's success testing, I made following;
1) I calculated the Ah for lower top voltage and saved it to 'Current & Voltage Monitoring' and 'Charge/Discharge configuration'.
2) Set new values for 'Battery charge voltage' and 'Fully charged voltage'
3) I programmed the inverter to make forced charging, but it do not start charging cycle.
4) Refressing Charge/Discharge configuration page give always two notes
  a) Discharging is prevented
  b) Charging is prevented
5) Those checkboxes are not checked
6) Made a connection to DiyBMS from another computer (if it is browser problem), but same notes will occur.
7) Tried check and uncheck above settings >> 'Settings saved' 

Now, the battery voltage is a bit lower than 'Battery discharge voltage minimum' setting.

What could cause the notes at Charge/Discharge configuration page ?

stuartpittaway, 2023-03-31:
The rules page will also trigger error messages to the inverter, like low cell voltage or low/high bank voltage.

Check there are no rules active (highlight in green)

closed 2023-04-02 by deekoowee