stuartpittaway/diyBMSv4ESP32 #193: no PWM on module board for discharge fet

Bingo2023, 2023-03-30:
As you see on the picture I have an 32 cell setup. I use a cell-board design for big LifePo4 cells. The setup & cabling is quite clean w/ short wires an min. connectors.

Boards run at 9.6 kboud.
Cycle time is 1720 ms (for 32 modules).
O.O.S errors: 3 (10 days uptime)
CRC Errors: 0
SW Version: V440_COMMS_9K6 (7d958c9f)

As soon as balancing starts communication breaks down completely. I can clearly see, communication stops at first module witch is balancing.
This is kind of dangerous because we are already at upper voltage limit and than BMS controller is total blind (no connection -> no voltage).

I agree that situation becomes worth for this design because with bigger cells u go for bigger balancing currents and that cause more disturbances. Layout is pretty clean but communication lines have to cross current path and pickup disturbances. But this design is best&clean for those kind of cells and it avoids lots of extra lose cables that potentially cause short circuit by rough handling.

I suggest that we change module SW and not use any PWM for controlling the discharge fet. For many reasons we should avoid switching signals with some amps on unshielded PCBs.
I think a normal 2 point temperature control would be enough to control/protect module boards from over heating.

[image]

stuartpittaway, 2023-03-31:
You'll find that running the boards at 5k Comms speed helps with this a lot.  As you mentioned there is significant voltage drop when the pwm is running.

Bingo2023, 2023-03-31:
OK I will try 5k com speed.

Even if it helps, I think the reasons I mentioned are valid, why not use a 2 point Temperature control for that?
That would improve stable communication, and reduce EMC issues.

stuartpittaway, 2023-04-01:
Yes, I'll take a look at a simple two point control, but I was trying to get you a quicker improvement

Bingo2023, 2023-04-01:
> Yes, I'll take a look at a simple two point control, but I was trying to get you a quicker improvement

and I really appreciate your fast response and the tremendous work you spend here.

stuartpittaway, 2023-04-12:
Just picking up this issue again - using a on/off temperature limit rather than PWM should be easy to implement.

I would suggest using the "Bypass over temperature" value as the maximum (normally 65°C), and then once that is reached, wait for a 20% drop - to 52°C before enabling balance current again.

Would this make sense to you?

If we use a fixed "20%" value, I don't need to amend the controller code much to gather a new "lower" temperature from the user.

Bingo2023, 2023-04-12:
That sounds like a good solution. I agree fixed 20% should be OK - we don't need to make it too complicated.

Bingo2023, 2023-07-21:
@stuartpittaway is this already in some release implemented?
I saw you created quite a lot new versions :)

stuartpittaway, 2023-07-21:
No not yet!