stuartpittaway/diyBMSv4ESP32 #194: Controller stopped sending CAN messages

deekoowee, 2023-04-02:
Hi @stuartpittaway 
Installed firmware 4e72db94af375c391bad47b9b98507b3271b069d
The CAN messaging stopped;
1) Battery SoC was over 100% (~140%, while charging, 'Override State of Charge' was checked)
2) Discharging battery
3) When battery SoC reached 100%, 'CAN sent:' stopped counting.
I catch it, because inverter complained missing battery.
However discharging continued.
I reset controller and 'CAN sent' counting started again and inverter was happy again.

stuartpittaway, 2023-04-02:
I've not seen that behaviour so thanks for reporting it.

I'm concerned your inverter did stop discharging if it lost communication with the BMS?

deekoowee, 2023-04-02:
Inverter did not stop discharging, when communication to BMS was lost

deekoowee, 2023-04-16:
Hi, same problem occured again. No CAN messages to inverter.
Inverter continued discharging.
When restarting the controller, screen shows a note 'Failed to save settings'
Firmware version is same: 4e72db94af375c391bad47b9b98507b3271b069d

stuartpittaway, 2023-04-17:
Hi @deekoowee nobody else has reported an issue with CAN comms stopping - this could be either:
1. A problem with the inverter/charger you have, which stops receiving the messages
2. A potential faulty ESP32 and/or CAN chip on the controller PCB

I've been running the CAN code for over 6 months now, most recently for 70+ days straight, with over 31 million CAN messages sent!  I'm using a SOFAR ME3000SP inverter.

[image]

deekoowee, 2023-05-16:
Hi @stuartpittaway 
I have continued the testing with another controller with new ESP32.
I purchased Seeed Studio's USB-CANBUS logger and now  I have canbus log for this issue if it helps.

Today, the new controller stopped sending the canbus messages.
Attached the canbus log file, which shows that there's no messages from controller after 19:37:08:772. 
When CAN communication lost, There's no other messages than inverter ID 0x305 (can be seen at almost end of canbus log file).

To make log smaller, I have cut unnecessary lines from beginning of the log.
[CANBUS log 20230516_190333.txt](https://github.com/stuartpittaway/diyBMSv4ESP32/files/11490744/CANBUS.log.20230516_190333.txt)

Time of inverter's log is about 00:01:43 ahead to canbus log.  Time difference is my good guess, when comparing the point at canbus log, where inverter started to send ID 0x305.

Inverter's log have;
1) 19:26:40 The battery asked for a forced charge. #196 
2) 19:38:53 Battery CAN communication lost.

Inverter send ID 0x305 at 19:37:10:371 and 19:37:10:425
Could two ID 0x305 with in 0,1 sec period cause the controller crash (Does the controller read 0x305) ?

Normally ID 0x305 are with ~10sec period except 19:34:41:888 to 19:37:10:371 and 19:37:10:371 to 19:37:47:171

Could there be a problem in data between 19:34:41:888 and 19:37:10:371 , causing the inverter to not sending ID 0x305 ?
I mean, if data is making the inverter 'crazy'.

I have also the voltage curve from HA, which shows 'wild' at 19:25:30 (almost same time, when inverter's log have 'The battery asked for a forced charge.' 
[image]

Because the inverter didn't get any messages from controller, it continued the battery discharging and start charging the battery after controller reset, when canbus started again.

deekoowee, 2023-05-17:
Battery was discharging.

Here's the current graph from same period
[image]

I think, I can log the serial console output directly to file with Putty.
I'll put it online together with canbus logger, then I have to wait.

deekoowee, 2023-05-24:
Hi @stuartpittaway 
Here's the serial log.
I cutted it from the original log file.
First error can be found in line 1289

[serial log.txt](https://github.com/stuartpittaway/diyBMSv4ESP32/files/11553410/serial.log.txt)

stuartpittaway, 2023-05-26:
Thanks for this @deekoowee.   I've so far, not received any other reports of this bug.

I'm wondering if @atanisoft could shed any light on this CAN issue?

deekoowee, 2023-05-30:
Hi,
Yes, I can make testing for that version.
Do you need canbus & serial logs ?

stuartpittaway, 2023-05-30:
The serial logs would be useful to see what errors occur, and how the controller deals with them.  Thanks

deekoowee, 2023-05-30:
The controller has updated with next version code and logging is activated.
Now just wait ...

deekoowee, 2023-05-31:
Here's the serial log

[serial log.txt](https://github.com/stuartpittaway/diyBMSv4ESP32/files/11614602/serial.log.txt)

stuartpittaway, 2023-05-31:
Thank you @deekoowee 

So the code I added detected that the CANBUS went into an error state (recovery needed).  The code automatically attempted to recover (```Initiating recovery```) however it didn't appear to work, and the CANBUS entered the STOPPED state.

@atanisoft is recovery normally possible?  Once in STOPPED state, can I re-start ?

```
D (44203473) diybms: Sent CAN message 0x351
D (44203494) diybms: Sent CAN message 0x355
D (44203514) diybms: Sent CAN message 0x356
E (44203534) diybms: Fail to queue CANBUS message (0x103)
I (44203534) diybms: CAN STATUS: rx-q:0, tx-q:0, rx-err:0, tx-err:128, arb-lost:13, bus-err:44, state: OFF / RECOVERY NEEDED
W (44203539) diybms: Initiating recovery
E (44203562) diybms: Fail to queue CANBUS message (0x103)
I (44203562) diybms: CAN STATUS: rx-q:0, tx-q:0, rx-err:0, tx-err:0, arb-lost:13, bus-err:44, state: STOPPED
E (44203585) diybms: Fail to queue CANBUS message (0x103)
I (44203585) diybms: CAN STATUS: rx-q:0, tx-q:0, rx-err:0, tx-err:0, arb-lost:13, bus-err:44, state: STOPPED
E (44204589) diybms: Fail to queue CANBUS message (0x103)
I (44204589) diybms: CAN STATUS: rx-q:0, tx-q:0, rx-err:0, tx-err:0, arb-lost:13, bus-err:44, state: STOPPED
E (44204613) diybms: Fail to queue CANBUS message (0x103)
I (44204613) diybms: CAN STATUS: rx-q:0, tx-q:0, rx-err:0, tx-err:0, arb-lost:13, bus-err:44, state: STOPPED
E (44204636) diybms: Fail to queue CANBUS message (0x103)
I (44204636) diybms: CAN STATUS: rx-q:0, tx-q:0, rx-err:0, tx-err:0, arb-lost:13, bus-err:44, state: STOPPED
E (44204659) diybms: Fail to queue CANBUS message (0x103)
```

atanisoft, 2023-05-31:
> is recovery normally possible? Once in STOPPED state, can I re-start ?

Yes, it should be possible. The call to `twai_initiate_recovery` will start the recovery (as you have seen) and it seems instead of returning to a running state like the old CAN driver did it goes into a STOPPED state. You can add one more `else if` clause for `TWAI_STATE_STOPPED` and call `twai_start` and then try sending the message.

You could also use the alerts mechanism to have it tell you when certain events occur so you can initiate recovery / restart / etc. I've shifted to the lower level TWAI HAL APIs and when `TWAI_HAL_EVENT_BUS_RECOV_CPLT` is raised I immediately trigger the bus restart.

deekoowee, 2023-06-01:
Controller shows this;

[image]

What does it mean ?

stuartpittaway, 2023-06-01:
> Controller shows this;
> What does it mean ?

This is a count of errors for RECEIVING from the inverter - the "heart beat" which the inverter sends the controller about every 10 seconds.  This error counter is incremented when a message isn't received (such as when the CANBUS is stopped/in error).

stuartpittaway, 2023-06-01:
@deekoowee thanks for trying out the various attempts to get this working for you.  Heres the latest one.

https://github.com/stuartpittaway/diyBMSv4ESP32/suites/13292392686/artifacts/725613861

@atanisoft thanks for the help.

deekoowee, 2023-06-01:
Latest one uploaded to controller.
Logging on and just waiting...

deekoowee, 2023-06-05:
Controller shows this; 
[image]

4 errors and controller has passed them without a problem.

stuartpittaway, 2023-06-05:
Great, so it looks like the recovery on the CANBUS works.

I still think there is a software fault in the inverter, but if this helps then great.

closed 2023-06-06 by stuartpittaway