stuartpittaway/diyBMSv4ESP32 #196: Forced charging ?

deekoowee, 2023-04-06:
Hi,
I was squeezing very last mAh 's from battery.
When battery reached 'Battery discharge voltage minimum' setting, Inverter log says 'The battery asked for a forced charge'.
Inverter started to charge the battery.

I checked manually the 'Prevent charging', but seems that inverter still charges with ~0,6A.
What could cause the small charging current ?

stuartpittaway, 2023-04-06:
What state of charge did the battery get to? What inverter/charger?

My sofar inverter charges with 8amps if the SoC goes below 20%

deekoowee, 2023-04-11:
I made some 'easter' testing.
I run battery down to 14% SoC by 'Prevent charging' setting. (Minimum SoC can be adjusted from 15% to 80%)
'Prevent charging' did not complete stop the charging and that was confusing me.
Unchecked the 'Prevent charging' and inverter started to charge with ~9A. Charging continued to 15% SoC.
Inverter is Imeon 9.12

stuartpittaway, 2023-04-12:
Ok - the "prevent charging" and "prevent discharge" are values which are transmitted to the charger/inverter - if that device choses to ignore the BMS, it can.

So is there an issue to look at here?

deekoowee, 2023-04-14:
I looked the pylon_canbus.cpp and found the lines 38 and 39.
If I underestand correct, the limitation is set to 0,1A , not 0A ?
I don't know, where the rest 0,5A comes. Maybe measuring error ?

stuartpittaway, 2023-04-17:
I found if we transmit zeros to either Victron or PylonTech emulation - they treat the values as "don't exist" or worse, they immediately start to discharge the battery at 100% !

What value is the maximum in the "dyn. charge current" box, when the 0.6Amp is charging ?

[image]