stuartpittaway/diyBMSv4ESP32 #199: new concept for battery protection

Bingo2023, 2023-04-12:
I would also like to contribute something to this DIY project as I'm using it for my own DIY battery (48V 400Ah; 16S2P [one controller w/ 32 module boards]).

On the attached picture you can find a new concept that would be little more forward towards PV battery usage.
The Rules page and Rules logic need to be adapted heavily but I think it would also simplify things and improve safety by making the config easy/better to understand.

Please join the discussion to improve this idea so that we have an straight forward concept that improves this project - if we decide to implement it.
[image]
[image]
[image]

red0909, 2023-04-13:
no this is not good 
it is good how it is now fully automatic, a manual reset of the controller is not usefull.
you have to disconnect the charger or the load seperatly this is the task of a bms.
if you have to disconnect a main contactor you can do so now without problems you need only a precharge circuit for the connection again

i can understand if someone need a additional precharge relay rule but the other rules should stay as they are

atanisoft, 2023-04-13:
> it is good how it is now fully automatic, a manual reset of the controller is not usefull.

I agree, having to reboot the BMS to reset relay state is a major regression that shouldn't be introduced to the code base.

However, I do agree some of the proposed features / rearrangement of the rules makes some sense. There is one caveat to that though, not everyone is using Victron / Pylontech / etc protocols. The proposal should include compatible functionality for "dumb" chargers that are relatively inexpensive and in use by a number of users. I have proposed to @stuartpittaway some time ago adding support to talk to EPSolar / EPEver devices via RS-485 but it would have been a bit out-of-scope for this project at the time, though some of the features make sense now (such as turn on / off charger or inverter via RS-485 packet).

Bingo2023, 2023-04-16:
> I agree, having to reboot the BMS to reset relay state is a major regression that shouldn't be introduced to the code base.

it is not an regression but an major improvement in safety. All of there marked rules are related to Cell protection witch should lead in final step to open the main contactor (the only way for BMS to protect the cells).

Let me make it more clear: "manual reset" means power reset or use the little switch on the ESP32 board OR do a software reset via the Webinterface (today: "More" -> "Utilities" -> "Restart Controller").

Something like this is required for the BMS, today I do it with setting the "reset value" to a number witch is never reached after a rule is activated, but this is not failesafe as u make easy mistakes and lots of people don't understand.

Let me give you an example why an manual acknowledge of the user is mandatory:
Imagine BMS is running stable for 2 years - after some time you never check it as it is always fine.
After 2 years your battery charger is doing an auto softwareupdate that increases max charge voltage +2V (Softwarebug, **first failure**). Now your BMS will protect the battery by opening the main contactor while charging with high current. After the over voltage situation disappears the BMS rule gets reset automatically and most likely the charging will continue straight away (depending on system design, latest next day is happens again). The main contactor can do "open under high current" only very few times and breaks down in most cases in "always on". Next charge cycle you end up in the ultimate catastrophe.
As you see with this little (realistic scenario - in my opinion) you don't have a chance to react to it. The faulty Softwareupdate could be a pulled or interrupted wire as well.
According to ISO 26262 you should only consider single faults, no double faults.

Bingo2023, 2023-04-16:
> The proposal should include compatible functionality for "dumb" chargers that are relatively inexpensive and in use by a number of users.

I think that is a good idea that we can include in the new proposal. Can you give me some more details what is needed for that?
I guess we need a rule to switch a Relais according to the overall pack voltage, anything else needed?

red0909, 2023-04-16:
> Now your BMS will protect the battery by opening the main contactor while charging with high current. After the over voltage situation disappears the BMS rule gets reset automatically and most [...]

this happens only without communication
so you need a better inverter with canbus communication, then there will be no problems. the charge voltage comes from the bms and it is dynamic if one cell rises more, the bms lowers the charge voltage.

so the rules are only for emergency and it is not good to turn the bms manualy on it should do it by itself this is the purpose of a bms.

i you want to disconnect your battery set a very high reset value, disconect via bi stable relay that you can manual connect via a push switch again if you want.

Bingo2023, 2023-04-16:
> so the rules are only for emergency and it is not good to turn the bms manualy on it should do it by itself this is the purpose of a bms.

the purpose of a BMS is to protect the cells. The only safe state for that is disconnected (-> main contactor open).
The BMS will also reconnect the cells AFTER acknowledge of the user.

Imagine the above scenario with the Softwarebug being a misinterpreted "charge voltage" coming via CAN bus. Same result.

red0909, 2023-04-16:
> misinterpreted "charge voltage" coming via CAN bus. Same result.

well and what if the rules are misinterpreted? or the relay coil does not work propperly? and what if a lightning strikes?  what if....
you cannot cover all scenarios

for your scenario i have a second bms on each of my batteries that does the disconnect if diy bms is not working.

red0909, 2023-04-25:
> You never want the BMS to shut off, and prefer damage to the cells rather than damage to the boat and/or humans!

maybe you can add a connector to the controller for emergency in future releases. to overide the rules and force rules off.

stuartpittaway, 2023-09-12:
Going to close this issue as I don't think any further work will be made on this topic.

closed 2023-09-12 by stuartpittaway