stuartpittaway/diyBMSv4ESP32 #202: SD card disappeared

deekoowee, 2023-04-14:
'Mount' button do not mount the SD card. Complains at right top corner of screen 'Request failed'. Same note appears, when refreshing (F5 or Ctrl-F5) browser. At yestarday, the SD card was ok.
Rebooting the controller didn't help. Also tried unpower / repower the controller.
Controller seems to work ok without SD card and SD card is readable with computer.
Replaced the display module, where the SD card slot is. Fault wasn't in display module.
Replacing the ESP32 didn' help.
Also found, that Flash memory functions has stopped to work. Can't save the configuration or upload file.
Ideas ?

**Hardware/Software Versions**
Controller version (from PCB): 4.2
Processor: ESP32
Version: [0f10cbb1d637bb0de7bf4248b9982970bceb554c](https://github.com/stuartpittaway/diyBMSv4ESP32/commit/0f10cbb1d637bb0de7bf4248b9982970bceb554c)
Compiled: 2023-04-13T10:28:59.928Z

deekoowee, 2023-04-15:
Solution; switch back to release 4e72db94af375c391bad47b9b98507b3271b069d

stuartpittaway, 2023-04-17:
This bug is caused by an incorrect reply from the API service - ```null)``` should be ```null``` in the /api/storage
Request Method: GET

commit 5b7135f8 (2023-04-17): Fix #202
closed 2023-04-17 by stuartpittaway