stuartpittaway/diyBMSv4ESP32 #204: Discharging limited to 49V ?

deekoowee, 2023-04-15:
I have tracked the charging / discharging curves for 12 days and found, that discharging stops always to ~49V. and it is not going lower.
My settings:
[image]

At a beginning, I have set the discharge limit to 49V, but have set it over week ago to 46.2 V as above settings shows.
What could cause the 49V limit ?

**Hardware/Software Versions**
Controller version (from PCB): 4.2
Processor: ESP32
Version: 4e72db94af375c391bad47b9b98507b3271b069d 
Compiled: 2023-03-29T08:22:59.136Z

stuartpittaway, 2023-04-15:
What limits are set on the charger/inverter?

deekoowee, 2023-04-15:
Inverter limits are 46 - 55.8 V
When battery voltage drops to ~49V, inverter log says;  "The battery asked for a forced charge."

stuartpittaway, 2023-04-17:
Discharging will also stop when the cell limit is reached "Individual cell min. voltage (mV)" - that would need to be lower than 2880 to work in your scenario - default is 3050.

deekoowee, 2023-04-22:
I'm not sure, but does the 2880 mean 16S LiFePo battery (16 * 2880 = 46080) ? 

My battery is 14S Li-Ion and that's why I have "Individual cell min. voltage (mV)" 3300 (14 * 3300 = 46200)
3300 is already little a bit under the charge knee.
I tried "Individual cell min. voltage (mV)" of 3000 and 2900, but lowest discharge voltage did not go lower than ~49V

deekoowee, 2023-04-24:
Inverter/charger discharge limit is set to 46V.
When battery voltage drops to ~49V, inverter receives message from DiyBMS "The battery asked for a forced charge."
But inverter doesn't start forced charge, because ~49V is more than lowest voltage 46V

deekoowee, 2023-04-25:
"Force low State of Charge" is not checked.
I'm running inverter with latest firmware.
SoC is around 52%, when the inverter log informs "The battery asked for a forced charge."
There's only the inverter and DiyBMS connected together with canbus, no other devices in canbus.

stuartpittaway, 2023-04-26:
Ok, that is very strange - I was expecting you to say the SoC had fallen below 20%.

I'm not really sure what is causing that issue.

deekoowee, 2023-04-30:
Ínverter is Imeon 9.12

I used the MQTT to get values to Home Assistant and graphs shows at the point of "The battery asked for a forced charge."
"Dyn. Chg. V" = 53,8V
"Dyn. Chg. Cur" = 160A

stuartpittaway, 2023-04-30:
Ok, very strange as the BMS isn't asking for a low voltage. Was the state of charge higher than 20% as well?

deekoowee, 2023-05-01:
Yes, SoC was more than 20%.

stuartpittaway, 2023-05-02:
This still feels like a problem with the software in the inverter, rather than diybms. Happy to be wrong though!

I don't have any other suggestions for you, CANBUS logging is probably a good idea, but diybms never requests forced state of charge.

deekoowee, 2023-05-02:
I play with Charge/Discharge configuration check boxes and inverter log says ; 
"The battery asked for a complete charge to rebalance cells voltage."
But inverter didn't start charging.

stuartpittaway, 2023-06-23:
Hi @deekoowee sorry only just noticed you updated this ticket.

The two messages will appear automatically if the charging parameters are exceeded - for example over voltage or temperature.  The controller will ask the charger to stop.

They should only be transient though, and show for a few seconds before being cleared and normal operation resuming.