stuartpittaway/diyBMSv4ESP32 #205: Module Configuration not working in Release-2023-04-13-10-32

Linusten, 2023-04-15:
**Describe the bug**
Module configuration is not working in the firmware release 2023-04-13-10-32.

**Hardware/Software Versions**
Controller version (from PCB): 4.5
Processor: ESP32
Version: [0f10cbb1d637bb0de7bf4248b9982970bceb554c](https://github.com/stuartpittaway/diyBMSv4ESP32/commit/0f10cbb1d637bb0de7bf4248b9982970bceb554c)
Compiled: 2023-04-13T10:28:59.928Z
Module Version: 4.5

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Modules'
2. Click on 'Configuration'
3. See error
"Cannot communicate with the controller for status updates"
Http response on  http://XXX/api/modules?c=0 --> Server has encountered an unexpected error
Previous release of the controller firmware is working fine.

maciejkrasuski, 2023-04-15:
Confirm - have the same issue.

stuartpittaway, 2023-04-17:
Bug introduced in last release - also reported in #203 

Fixed in 580467c1897337199647abccb1d114ca6de0d3dd