stuartpittaway/diyBMSv4ESP32 #206: charging algorithm needs to be adjusted

buddhafragt, 2023-04-17:
I am currently testing the controller v4.4.3 and it is already a big improvement in terms of Victron integration.
However, I think the charging algorithm needs to be adjusted:
Let me summarize the current state:
The DIYBMS charges the battery with a customizable current/voltage curve up to the “Charge target cell maximum voltage” value which is then maintained depending on the available charging power.
However, this is not an optimal solution for how LiFePo4 batteries should be charged. A very good post that explains the problem very well can be found here:
[Nordkyn Design - Marine Design & Engineering – 21 Feb 21](https://nordkyndesign.com/charging-marine-lithium-battery-banks/)
[Charging Marine Lithium Battery Banks | Nordkyn Design 1](https://nordkyndesign.com/charging-marine-lithium-battery-banks/)

In short, the problem is that the battery should not be kept constantly at e.g. 3.5V, but charged and balanced to 3.5V and then discharged again to e.g. 3.33V. Keeping the battery permanently at 3.5V will damage the cells. (please read the link to this topic).

Victron has integrated this charging algorithm in its MPPTs, but unfortunately this is completely deactivated by activating DVCC and the DIYBMS takes control. So the charging algorithm implemented by Victron should also be integrated in the DIYBMS!

unfortunately, there is also no longer a possibility to set the charging current during balancing, this should be added again.

maciejkrasuski, 2023-04-17:
This is exactly what I was talking about on my post on facebook group. Thanks for the articles.

stuartpittaway, 2023-04-17:
@buddhafragt great reading thanks for the links.

What would you propose as a solution?

We could do something similar to the SoC 100% reset calculation, where the voltage is > X and current < Y for Z minutes ?

However, that isn't always perfect - if you consider solar charger for instance, it may have a very low current in low sunlight.

Alternatively, we could use the SoC (but has similar issues to above).

Perhaps using the highest cell voltage and current?

Once we know the "trigger" it should be easy enough to set the charge voltage to a "float" level (3.3V/cell) for a period of time, or alternatively, until the SoC drops X percent ?

maciejkrasuski, 2023-04-17:
@stuartpittaway if I could join to the discussion because it is really interesting ;) I'm thinking about this tail current and the low sunlight on cloudy day. To charge battery to certain voltage you need some energy so if there will be very low current from the charge controler there voltage won't hit the target voltage. And if it will be on the last stage of charging - the current is already very limited by your dynamic charge algorithm so I don't know if this will be the issue. If you add these parameters to config we will test and find the sweet spot for that.
Or maybe you can add fourth parameter that count the current flow in both way and count the SoC on the energy transfered to and from the battery with some additional parameter for losses.
About when restart charging I think that configurable SoC percent will be great.

N1c084, 2023-04-17:
I think using the highest cell voltage for 100% reset is the best .

But you need to do a mean of each cell (calculate soc of each cell and do a
mean) because when the balancing is not good the other cells do not have
the same soc so the global real soc can be less than 100%.
> when the first cell reach the highest value for x seconds > reset soc to
100%> adjust the value regarding mean value of all cell in serial

For the "trigger" the SoC drops X percent is the best solution , floating
on lion is not useful and can be dangerous because the risk of over voltage
on the higher cell

stuartpittaway, 2023-04-17:
> calculate soc of each cell and do a mean

That isn't possible - you have no way to calculate the SoC on each cell.  The whole bank has a SoC because of amp-hour counting - but even that is not perfect.

stuartpittaway, 2023-04-17:
> To charge battery to certain voltage you need some energy so if there will be very low current from the charge controler there voltage won't hit the target voltage.

The more I think about it, I agree, if solar charger is limited in current, then the voltage it outputs will have to drop to compensate.  So if the battery voltage is already near "max" then the cells will be full regardless.

stuartpittaway, 2023-04-19:
So to consolidate the above ideas, how does this sound:

1. Ignore the individual cell voltages - that's the job of balancing.
2. When the SoC hits 100% fully charged, start a timer for X minutes - this allows top balancing to continue, probabably 30 or 60 minutes.
3. After the timer is elapsed, set charging voltage to a "float" level (LIFEPO4=3.3V/cell) and hold at this level
4. Once SoC drops below Y % (say 98%), drop out of "float" and return to normal operation

All the above would be user definable values.

stuartpittaway, 2023-04-19:
When I mentioned the word "float" I deliberately used speech marks to highlight that I didn't really mean "float" ! Probably got lost in translation!!

I know lithium can't float, however, not all chargers will stop charging, and additionally, you don't want to shut down solar PV - as that may also be powering other loads during the day (on a boat or house for example).

So dropping to a lower level voltage after the cells are full, allows other devices to continue to operate, but won't charge the cells as the voltage is very low and no current will flow.

red0909, 2023-04-19:
> So to consolidate the above ideas, how does this sound:

sounds good but the values of time, voltage and soc should be set by user. not fixed.
maybe you can also add the #181

buddhafragt, 2023-04-19:
> So to consolidate the above ideas, how does this sound:
>
>     1. Ignore the individual cell voltages - that's the job of balancing.
>
>     2. When the SoC hits 100% fully charged, start a timer for X [...]

Stuart, I think that's a good solution, that's kind of how the Vicron loading algorithm works. 
And I found, as written above, the possibility to limit the current to a certain value during balancing very useful!

MickSquared, 2023-04-24:
Stuart,

Your proposed algorithm sounds great.

I know i'm late to the party but i would like to add another twist.

I would like to be able to dynamically adjust the CCL to a lower value once one or more cells have started to balance.

This would allow charging to continue longer but enable the limited balancers to stay in the game longer.

Like your proposal they should be user adjustable and would have two values.
e.g. within DIYbms we could have the following values and the correct one be published to CAN bus based on balancing
CCL_FULL_RATE = 70  (amps)
CCL_BALANCING_RATE = 25  (amps)

maciejkrasuski, 2023-04-24:
> Stuart,
>
> Your proposed algorithm sounds great.
>
> I know i'm late to the party but i would like to add another twist.
>
> I would like to be able to dynamically adjust the CCL to a lower value once one [...]

Hmmm but it's already done by Dynamic Charge Algorythm. It's not enough for you?

MickSquared, 2023-04-24:
> Hmmm but it's already done by Dynamic Charge Algorythm. It's not enough for you?

My bad - you are correct as it stands it should accommodate what i was looking for.

This week-end i will update everything to latest code be it Victron, Controller and modules and have another go.

At present everything is getting quite dated.

Stuart - sorry to waste your time - feel free to close/delete my point whatever suits you.

buddhafragt, 2023-07-29:
what is the "all in one" branch?

stuartpittaway, 2023-07-29:
> what is the "all in one" branch?

This...  https://youtu.be/RwrlYUxnfZ0

stuartpittaway, 2023-09-12:
Closing this as latest release now includes "float" function

closed 2023-09-12 by stuartpittaway