stuartpittaway/diyBMSv4ESP32 #207: Error messages overlap

maciejkrasuski, 2023-04-17:
**Describe the bug**
If you have lots of errors and warnings its hard to read it because popups constantly shows above this red alert bar. The alerting system should be unified to one type or popups should be placed on the right corner rather than left like it is right now.

**Hardware/Software Versions**
Processor: ESP32
Version: [0f10cbb1d637bb0de7bf4248b9982970bceb554c]
Compiled: 2023-04-13T10:28:59.928Z

**To Reproduce**
Steps to reproduce the behavior:

It will be hard to reproduce, because I have problem with communication so there was some kind of error on the red bar and a lot of warnings in the front of it and when I closed the warning to see what is on the red bar it changed to sth else and new warnings new warning appeared.
[image]

stuartpittaway, 2023-04-19:
Thanks for reporting @maciejkrasuski.  Just for information, you can click on the "discharging is prevented" style messages and they disappear.

stuartpittaway, 2026-11-14:
Fixed in 6a7bdde on `March2026-MinorChanges`.

Notification pop-ups (warnings, errors and confirmations) no longer appear in the top corners over the red error bars. They now all stack in the centre of the screen, so the error bars stay readable. You can still click a pop-up to dismiss it, and they hide themselves after a few seconds as before (warnings after 15 seconds).

This will be in the next release once the branch is merged.

closed 2026-11-14 by stuartpittaway