stuartpittaway/diyBMSv4ESP32 #209: Victron CAN-bus voltage alarm swapped

bertvaneyken, 2023-05-04:
I think the voltage alarm is swapped in the Victron CAN bus implementation, can someone else have also a look at this code?

Line 296 

    //(bit 2+3) Battery low voltage alarm
    data.byte0 |= ((rules.rule_outcome[Rule::BankOverVoltage] | rules.rule_outcome[Rule::CurrentMonitorOverVoltage]) ? BIT23_ALARM : BIT23_OK);
    //(bit 4+5) Battery high voltage alarm
    data.byte0 |= ((rules.rule_outcome[Rule::BankUnderVoltage] | rules.rule_outcome[Rule::CurrentMonitorUnderVoltage]) ? BIT45_ALARM : BIT45_OK);

I couldn't explain the bank undervoltage alerts on my GX device when I was at the top end balancing as it overshot the values.
This might be the source of this error.

Maybe an improvement to add:
Add a option to set a specific charge current when cells start to balance, the setting available now to stop charging momentary when a cell starts dumping current.  This slows the process down a bit.

bertvaneyken, 2023-05-15:
@stuartpittaway  did you obtain documentation on the DVCC/Victron CAN-bus implementation or was it reverse engineered?

stuartpittaway, 2023-05-15:
> @stuartpittaway did you obtain documentation on the DVCC/Victron CAN-bus implementation or was it reverse engineered?

Victron provided some documentation for me, although there were some inaccuracies in it.

bertvaneyken, 2023-05-23:
I didn't notice the controller unexpectedly rebooted, when this happend the Victron Multiplus 2 gives a "low battery voltage" alarm and shuts the inverter off.

I'm not sure but if I change parameters/rules the software isn't happy after a while and reboots.

stuartpittaway, 2023-05-24:
Ok - in which case I would need to get a copy of the log files from the USB port on the ESP32 when this scenario occurs - unfortunately this would mean leaving a computer running to capture this output.

virtuvas, 2023-07-18:
@bertvaneyken 
check what I and others posted in #220:

https://github.com/stuartpittaway/diyBMSv4ESP32/issues/220#issuecomment-1639707646

update multiplus II firmware to latest and check again

V.

bertvaneyken, 2023-07-27:
Thanks for the tip!

I just upgraded my GX from 2.93 to 3.02 and the MPII from 5.02 to 5.05

virtuvas, 2023-07-29:
> 

please let us know how it's going!

stuartpittaway, 2023-09-12:
All - can this issue be closed now?