stuartpittaway/diyBMSv4ESP32 #210: Victron CANBUS - BMSError - possible "zero volt" from module error?

stuartpittaway, 2023-05-11:
As reported by Jochen Schmidt

Linusten, 2023-05-21:
Got this "Internal error alarm: Alarm" also a few times, dont really know what is causing the issue....

JochenSchmidt, 2023-05-21:
Before 9 days I installed a tiny old Eee PC with PuTTy logging the ESP32 USB stream.

But since then there hasn't been another internal error in the diyBMS.

I'll keep the Eee PC logging.

JochenSchmidt, 2023-05-21:
> Got this "Internal error alarm: Alarm" also a few times, dont really know what is causing the issue....

Did you also have a shutdown from the Multiplus / Quattro?

JochenSchmidt, 2023-05-21:
@Linusten 

> Got this "Internal error alarm: Alarm" also a few times, dont really know what is causing the issue....

Did you also have a shutdown from the Multiplus / Quattro?

Linusten, 2023-05-21:
Yes, got the shutdown everytime the Error occurs.

stuartpittaway, 2023-05-22:
Ok, definitely needs fixing, but need to narrow down the problem first - and the logs are the only way to achieve that.

JochenSchmidt, 2023-05-22:
@Linusten 
Do you have the possibility to log the ESP32 via USB with PuTTy?

[image]

JochenSchmidt, 2023-05-22:
[image]

[image]

JochenSchmidt, 2023-05-22:
Damn it, a few weeks ago the error came up very often and now that I've logged it doesn't anymore.

JochenSchmidt, 2023-05-22:
The only difference now is that the ESP32 and the 4.5 controller are now supplied with power via the USB port of the ESP32, by the small EeePC. Previously via the terminals on the 4.5 controller via a Meanwell power supply with 5V.