stuartpittaway/diyBMSv4ESP32 #211: Esp32 stopping after 1-4 days

Michi202020, 2023-05-19:
Hello Stuart, 
My system withe your board work 1 years very good (board. 4.5). 
I will check the new version from 17.4.2023 but after 1-4 days he stop 
The work. No touch panel or led blinking. 
After this I install 29.3.2023. It's a little bit better but it also stay after 3 or 5 days. 
Have you or other people's problem withe this? 

Many Thanks and best Regards 
Gärtner

stuartpittaway, 2023-05-19:
I've not seen this problem, my system is running over 100 days now without any reboot.

Do you have another esp32 you can try?

Michi202020, 2023-05-19:
I will check it tomorrow on a other battery system. I have 5 pieces from this system by my friends.

I

Michi202020, 2023-05-27:
Hello Stuart I have check withe 3 different esp32 but all of them stop after 1-2 days.
My Batterie is not connected to a inverter. I will test the communication to shunt and the cells Version 4.4. If I start there all good. I see the voltage current and many more but he will stop. Is the problem if I don't have connect to can bus?
I also test the last 4 Version. 

Best Regards
Michael

stuartpittaway, 2023-05-30:
Hello, I think the only way to debug this problem is to get a copy of the log messages which are output over the USB cable from the ESP32.

Ideally we need to log those to a file so we can see what errors cause the problem.

Have you tried running it without the TFT screen attached?

Michi202020, 2023-06-02:
Hello Stuart,

I have check the Version 21.02.2023 is working good but after this version he stop.
I have install a Display but for testing i dont install the Victron. In your software i dont activate the can bus only the chunt and the Cells V4.4.

I sent you the log file for Version 29.03.2023. 
[
[output_2023-06-02_07-43-58.log](https://github.com/stuartpittaway/diyBMSv4ESP32/files/11632393/output_2023-06-02_07-43-58.log)
](url)

Michi202020, 2023-06-02:
[image]

Michi202020, 2023-06-02:
[image]

Michi202020, 2023-06-02:
_
[image]
_

Michi202020, 2023-06-02:
_
[image]
_

stuartpittaway, 2023-06-02:
Thank you for the log files.

There are two problems, the first is multiple errors trying to connect to the WIFI - is the router too far away?  Can the signal get through the box you have built?

```
E (39913667) diybms: Connect to the Wifi AP failed
E (39943667) diybms: Connect to the Wifi AP failed
E (39973669) diybms: Connect to the Wifi AP failed
E (40003669) diybms: Connect to the Wifi AP failed
E (40033671) diybms: Connect to the Wifi AP failed
E (40063671) diybms: Connect to the Wifi AP failed
```

The second problem is a "overflow" in the code.  However, nobody else is reporting this at the moment.
Can I ask how many cells are you running?

```
E (40067683) diybms: Packet Failed

***ERROR*** A stack overflow in task rxq has been detected.

Backtrace: 0x400841d1:0x3ffda1b0 0x4008e329:0x3ffda1d0 0x40092149:0x3ffda1f0 0x400900b9:0x3ffda270 0x4008e438:0x3ffda2a0 0x4008e3e8:0x3ffda1b0 0xfffffffe:0x3ffda1d0 |<-CORRUPTED

ELF file SHA256: afc35a9044bcdb18
```

Michi202020, 2023-06-02:
Hello Stuart,
I have install 12 cells serial and 3 paralel. It's 36 Cells.

Withe the wifi I make withe my smartphone a Hotspot and config the battery. After this I install for eyample at my brother and friend but there I don't connect withe her router.
On the most time it's better other people's come not on this interface.

Best Regards
Michael

Michi202020, 2023-06-02:
Hello Stuart,

i have it flashed and we will see over the weekend. For this error i need ca. 8 Hour.

Jestergnet, 2023-06-05:
What type of contactor do you use?

Jestergnet, 2023-06-05:
Michi202020, what type of contactor do you use?

Michi202020, 2023-06-05:
Jestergnet, The Batterie and the contactor are from ID3 Car. The Contactor TE EVC175.

[https://www.te.com/deu-de/products/relays-contactors-switches/contactors/automotive-contactors/evc-175-main-contactor.html](url)

Linusten, 2023-06-05:
> Great thank you - surprised you ran into those problems, and other people haven't.

I think maybe the issue  is related to : https://github.com/stuartpittaway/diyBMSv4ESP32/issues/210

stuartpittaway, 2023-06-06:
@Linusten I don't think its related to #210 but will keep investigating.

Looks like this problem is fixed, so I'll close the ticket.

closed 2023-06-06 by stuartpittaway