stuartpittaway/diyBMSv4ESP32 #213: Diy BMS data agregation via can bus. Multi Bank improvement

lorenzo6201, 2023-05-29:
I have 13 controllers and 182 cell monitors and is a pain to monitor all , I have to open 13 browser pages , remembering 13 ip`s , etc. 
Currently only one is talking to the inverter via can bus but is sending the incomplete data. if i have 100 A on discharging , this controllers reports only 7A because the current is divided among 13 banks. 
After reading a few forums , asking questions the DIY BMS code should be able to do the following : 
1. At boot every controller has to self assign a unique ID or manually assign a unique ID via web interface or dip switch. I prefer manual from web interface , eg controller 1 , 2 , etc. 
2. The web interface should have the option to set the controller as master or slave. 
3. The master controller should request and process data from the slaves via can bus or serial , like average voltage , lowest cell , highest cell , adding up the currents , average SOC , etc. 
4. If a slave pack has an issue , that controller should disconnect that pack from the rest without triggering a total shutdown of the system. This way , that pack can be removed from the rack and serviced without shutting down the system. 
5. The master controller should have a page to view the status of the slaves , minimal info like total voltage , highest cell , lowest cell , amp in and out , current draw , soc , etc and ip address for quick access. 

The system i own is not mission critical , but i prefer not to shutdown  the system when i remove a pack. 
Right now the rack has two positive copper busbars and 2 negative busbars. The rack is split in 2 , 7 banks on the right , 7 on the left. The total rack capacity is 88.3Kwh , divided into 14 packs , 6.3 Kwh each. 
As for safety features , the rack is equipped with flame sensor , smoke sensor , and independent temperature sensor. If one of the sensors is triggered a fire alarm would sound. The rack walls are made of 4 mm steel ,  with a heat resistant metal pipe venting outside. In the near future , the rack will be fitted with a water flooding sistem , triggered by the flame sensor. In case of open flame , the entire rack will be filled to the top with water. The rack is also equipped with a master relay that disconnects all packs from the inverters in case one of the sensors is triggered. 
  Each pack has a metal enclosure made of 1mm steel. I used only fire proof materials. each pack is equipped with a 100A relay driven by the DIY BMS controller but can be disconnected manually and in the future will disconnect automatically if master relay is triggered or the pack is removed from the rack. 
If daisy chain is a go i would add 15 rj45 connectors wired in parallel for can bus comms  so i can plug in each pack. 

If anyone is interested into developing this feature , i have the perfect test bench. I also have another setup made with a spare growatt sp5000ES , 7 controllers with 98 cell modules and 98 25ah lifepo4 cells ,  set up and ready to test new code. So there is no need to invest money into buying controllers , cells , inverters. I can also provide remote access via anydesk at any time , 24/7 , on a separate networked computer for debugging purposes. The computer would be with a fresh windows install , no personal data on it , separate internet provider with symmetrical 1Gb fiber optic internet.
I'm also available at the location during day time so if required i can assist with plugging programmers , rebooting , disconnecting or connecting stuff.

stuartpittaway, 2023-05-30:
Hi @lorenzo6201.  This isn't a topic I'm actively working on at the moment, however I know user ```menningd``` (https://community.openenergymonitor.org/u/menningd) was working on something similar.

lorenzo6201, 2023-05-30:
This would be the cherry on top. At least a skeleton code or something to start with. I`m sure there is someone here good at coding so we can work on that. I tried to look into your code but looks like sorcery to me. Like I said , I have the test bench , all we need is the code.