stuartpittaway/diyBMSv4ESP32 #214: INA 229 , changing from adc_range 1 to adc_range0 for 75mV shunts.

lorenzo6201, 2023-06-05:
Like I mentioned in the forum I would like to use 75mV shunts with ina229 addon board. I have tried to alter CurrentMonitorINA229.cpp and CurrentMonitorINA229.h 
It works but not precise. The value is higher or lower than real. I think I missed something or maybe is due to scaling to 50mV shunts. i looked several times over the code , used the TI ina229 calculator but I have no real idea where to put the values. 
I will copy paste the code here

CurrentMonitorINA229.cpp
`/*
 ____  ____  _  _  ____  __  __  ___    _  _  __
(  _ \(_  _)( \/ )(  _ \(  \/  )/ __)  ( \/ )/. |
 )(_) )_)(_  \  /  ) _ < )    ( \__ \   \  /(_  _)
(____/(____) (__) (____/(_/\/\_)(___/    \/   (_)

DIYBMS V4.0
INA229 CURRENT/ENERGY MONITOR CHIP (SPI INTERFACE)

(c)2023 Stuart Pittaway

COMPILE THIS CODE USING PLATFORM.IO
LICENSE
Attribution-NonCommercial-ShareAlike 2.0 UK: England & Wales (CC BY-NC-SA 2.0 UK)
https://creativecommons.org/licenses/by-nc-sa/2.0/uk/
* Non-Commercial — You may not use the material for commercial purposes.
* Attribution — You must give appropriate credit, provide a link to the license, and indicate if changes were made.
  You may do so in any reasonable manner, but not in any way that suggests the licensor endorses you or your use.
* ShareAlike — If you remix, transform, or build upon the material, you must distribute your
  contributions under the same license as the original.
* No additional restrictions — You may not apply legal terms or technological measures
  that legally restrict others from doing anything the license permits.

** COMMERCIAL USE AND RESALE PROHIBITED **
*/

#define USE_ESP_IDF_LOG 1
static constexpr const char *const TAG = "curmon";

#include "CurrentMonitorINA229.h"

void CurrentMonitorINA229::CalculateLSB()
{
    // Take a look at these for information on how it works!
    // https://dev.ti.com/gallery/view/4910879/INA228_229_237_238_239EVM_GUI/ver/2.0.0/
    // in above - click COG icon top left.
    // https://e2e.ti.com/support/amplifiers-group/amplifiers/f/amplifiers-forum/1034569/ina228-accumulated-energy-and-charge-is-wrong

    // 150A/50mV shunt =   full_scale_current= 150.00A / 50.00 * 40.96 = 122.88 AMPS
    //                     RSHUNT = (50 / 1000) / 150 = 0.00033333333
    //                     CURRENT_LSB = 150/ 524288 = 0.000286102294921875
    //                     R_SHUNT_CAL = 52428800000*0.000286102294921875*0.00033333333 = 4999.999 = 5000

    // 300A/75mV shunt =   full_scale_current= 300.00A / 75.00 * 40.96 = 163.84 AMPS
    //                     RSHUNT = (75 / 1000) / 163.84 = 0.000457763671875
    //                     CURRENT_LSB = 163.84/ 524288 = 2.648162841796875e-4
    //                     R_SHUNT_CAL = 52428800000*0.00057220458984375*0.00025 = 7499.9922688 = 7500

    // Calculate CURRENT_LSB and R_SHUNT_CAL values
    // registers.full_scale_current = ((float)registers.shunt_max_current / (float)registers.shunt_millivolt) * full_scale_adc;
    registers.RSHUNT = ((float)registers.shunt_millivolt / 1000.0) / (float)registers.shunt_max_current;
    registers.CURRENT_LSB = registers.shunt_max_current / (float)0x80000;
    registers.R_SHUNT_CAL = 4L * (13107200000L * registers.CURRENT_LSB * registers.RSHUNT);

    ESP_LOGI(TAG, "RSHUNT=%.8f, CURRENT_LSB=%.8f, R_SHUNT_CAL=%u", registers.RSHUNT, registers.CURRENT_LSB, registers.R_SHUNT_CAL);

    // Will the full scale current be over the 40.96mV range the ADC can handle?
    if (((float)registers.shunt_max_current * registers.RSHUNT * 1000.0) > full_scale_adc)
    {
        float true_max_current = ((float)registers.shunt_max_current / (float)registers.shunt_millivolt) * full_scale_adc;
        ESP_LOGW(TAG, "WARNING: True Max Current Measurable %.2f Amps", true_max_current);

        registers.CURRENT_LSB = true_max_current / (float)0x80000;
        registers.R_SHUNT_CAL = 4L * (13107200000L * registers.CURRENT_LSB * registers.RSHUNT);
        ESP_LOGI(TAG, "RECALC: RSHUNT=%.8f, CURRENT_LSB=%.8f, R_SHUNT_CAL=%u", registers.RSHUNT, registers.CURRENT_LSB, registers.R_SHUNT_CAL);
[... 912 lines cut ...]
    float Current()
    {
        // Current. Two's complement value.
        return -(registers.CURRENT_LSB * (float)readInt20(INA_REGISTER::CURRENT));
    }

    void ResetChargeEnergyRegisters()
    {
        // BIT 14
        // RSTACC
        // Resets the contents of accumulation registers ENERGY and CHARGE to 0
        // 0h = Normal Operation
        // 1h = Clears registers to default values for ENERGY and CHARGE registers
        write16bits(INA_REGISTER::CONFIG, registers.R_CONFIG | (uint16_t)_BV(14));
    }

    void SetINA229ConfigurationRegisters()
    {
        write16bits(INA_REGISTER::CONFIG, registers.R_CONFIG);
        write16bits(INA_REGISTER::ADC_CONFIG, registers.R_ADC_CONFIG);
    }
};
#endif
`
[image]

delboy711, 2023-08-11:
I have run into the same issue.  My 50A 75mV shunt did not read any current higher than 27A.  I finally realised the issue was the 40.97mV range set on the INA229 and not a problem with my inverter.

Looking at your code it is hard to work out what changes you have made, but you do not seem to have changed the ADCRANGE flag.  This is set in the function Configure()

From  line  ~489
```   
        //registers.R_CONFIG = _BV(4); // ADCRANGE = 40.96mV scale
        registers.R_CONFIG &= 0xffef ; //Set  ADCRANGE  to  163.84 mV scale
 ```

delboy711, 2023-08-11:
I have it working now for my 50A 75mV shunt.
The attached patch should work for any type of shunt. For 50mV shunts it sets the ADCRANGE to 40.96mV and for higher Voltage shunts it sets the range to 163.84mV.

The only change in the calculations is R_SHUNT_CAL is 4 times larger when the 40.96mV range is selected.file:///home/derek/Desktop/INA229.diff

```
diff --git a/ESPController/include/CurrentMonitorINA229.h b/ESPController/include/CurrentMonitorINA229.h
index 599f2ef..0557c01 100644
--- a/ESPController/include/CurrentMonitorINA229.h
+++ b/ESPController/include/CurrentMonitorINA229.h
@@ -293,7 +293,7 @@ private:
     float power = 0;
     float temperature = 0;
 
-    const float full_scale_adc = 40.96;
+    float full_scale_adc = 40.96;
     // const float CoulombsToAmpHours = 1.0 / 3600.0;
     const float CoulombsToMilliAmpHours = 1.0 / 3.6;

diff --git a/ESPController/src/CurrentMonitorINA229.cpp b/ESPController/src/CurrentMonitorINA229.cpp
index 32c1f9e..0ef51d5 100644
--- a/ESPController/src/CurrentMonitorINA229.cpp
+++ b/ESPController/src/CurrentMonitorINA229.cpp
@@ -36,32 +36,39 @@ void CurrentMonitorINA229::CalculateLSB()
     // in above - click COG icon top left.
     // https://e2e.ti.com/support/amplifiers-group/amplifiers/f/amplifiers-forum/1034569/ina228-accumulated-energy-and-charge-is-wrong
 
-    // 150A/50mV shunt =   full_scale_current= 150.00A / 50.00 * 40.96 = 122.88 AMPS
+    // 150A/50mV shunt =   full_scale_current= 150.00A / 50.00 * 40.96 = 122.88 AMPS ADCRANGE=1
     //                     RSHUNT = (50 / 1000) / 150 = 0.00033333333
     //                     CURRENT_LSB = 150/ 524288 = 0.000286102294921875
     //                     R_SHUNT_CAL = 52428800000*0.000286102294921875*0.00033333333 = 4999.999 = 5000
 
-    // 300A/75mV shunt =   full_scale_current= 300.00A / 75.00 * 40.96 = 163.84 AMPS
-    //                     RSHUNT = (75 / 1000) / 163.84 = 0.000457763671875
-    //                     CURRENT_LSB = 163.84/ 524288 = 2.648162841796875e-4
-    //                     R_SHUNT_CAL = 52428800000*0.00057220458984375*0.00025 = 7499.9922688 = 7500
+    // 300A/75mV shunt =   full_scale_current= 300.00A  ADCRANGE=0
+    //                     RSHUNT = (75 / 1000) / 300 = 0.00025
+    //                     CURRENT_LSB = 300/ 524288 = 5.72204589844e-4 = 0.000572204589844
+    //                     R_SHUNT_CAL = 13107.2e6 * 5.72204589844e-4 * 0.00025 = 1875
+
+    // 50A/75mV shunt =   full_scale_current=50.00A  ADCRANGE=0
+    //                     RSHUNT = (75 / 1000) / 50 = 0.0015
+    //                     CURRENT_LSB = 50/ 524288 = 9.53674316406e-05 = 0.0000953674316406
+    //                     R_SHUNT_CAL = 13107.2e6 * 9.53674316406e-05 * 0.0015 =    1875
 
     // Calculate CURRENT_LSB and R_SHUNT_CAL values
     // registers.full_scale_current = ((float)registers.shunt_max_current / (float)registers.shunt_millivolt) * full_scale_adc;
     registers.RSHUNT = ((float)registers.shunt_millivolt / 1000.0) / (float)registers.shunt_max_current;
     registers.CURRENT_LSB = registers.shunt_max_current / (float)0x80000;
-    registers.R_SHUNT_CAL = 4L * (13107200000L * registers.CURRENT_LSB * registers.RSHUNT);
+    registers.R_SHUNT_CAL = (13107200000L * registers.CURRENT_LSB * registers.RSHUNT);
+    if(registers.shunt_millivolt <= 50) registers.R_SHUNT_CAL * 4L;
 
     ESP_LOGI(TAG, "RSHUNT=%.8f, CURRENT_LSB=%.8f, R_SHUNT_CAL=%u", registers.RSHUNT, registers.CURRENT_LSB, registers.R_SHUNT_CAL);
 
-    // Will the full scale current be over the 40.96mV range the ADC can handle?
+    // Will the full scale current be over the range the ADC can handle?
     if (((float)registers.shunt_max_current * registers.RSHUNT * 1000.0) > full_scale_adc)
     {
         float true_max_current = ((float)registers.shunt_max_current / (float)registers.shunt_millivolt) * full_scale_adc;
         ESP_LOGW(TAG, "WARNING: True Max Current Measurable %.2f Amps", true_max_current);
 
         registers.CURRENT_LSB = true_max_current / (float)0x80000;
-        registers.R_SHUNT_CAL = 4L * (13107200000L * registers.CURRENT_LSB * registers.RSHUNT);
+        registers.R_SHUNT_CAL = (13107200000L * registers.CURRENT_LSB * registers.RSHUNT);
+        if(registers.shunt_millivolt <= 50) registers.R_SHUNT_CAL * 4L;
         ESP_LOGI(TAG, "RECALC: RSHUNT=%.8f, CURRENT_LSB=%.8f, R_SHUNT_CAL=%u", registers.RSHUNT, registers.CURRENT_LSB, registers.R_SHUNT_CAL);
     }
 
@@ -456,6 +463,8 @@ bool CurrentMonitorINA229::Configure(uint16_t shuntmv,
     registers.tail_current_amps = tailcurrent / 100.0;
     registers.charge_efficiency_factor = chargeefficiency / 100.0;
 
+    full_scale_adc = (registers.shunt_millivolt > 50) ? 163.84 : 40.96;     //Select ADC Range
+
     CalculateLSB();
 
     if (shuntcal != 0 && registers.R_SHUNT_CAL != shuntcal)
@@ -486,7 +495,7 @@ bool CurrentMonitorINA229::Configure(uint16_t shuntmv,
 
     // ESP_LOGI(TAG, "undercurrentlimit=%f, R_SUVL=%u", ((float)undercurrentlimit / 100.0F), registers.R_SUVL);
 
-    registers.R_CONFIG = _BV(4); // ADCRANGE = 40.96mV scale
+    registers.R_CONFIG =(registers.shunt_millivolt > 50) ? registers.R_CONFIG & 0xffef: _BV(4);  // ADCRANGE = 163.84mV or 40.96mV scale
 
     if (TemperatureCompEnabled)
     {
diff --git a/ESPController/web_src/default.htm b/ESPController/web_src/default.htm
index de58b13..b31e2aa 100644
--- a/ESPController/web_src/default.htm
+++ b/ESPController/web_src/default.htm
@@ -1395,7 +1395,7 @@
         <h2>Basic Settings</h2>
         <p id="b3">Ensure the current shunt parameters match the data sheet for your particular shunt resistor.</p>
         <p id="b4">
-          The DIYBMS current monitor uses a 40.96mV maximum scale, so shunt voltages over this will be scaled down
+          For 50mA internal shunts DIYBMS current monitor uses a 40.96mV maximum scale, so shunt voltages over this will be scaled down
           proportionally.
         </p>
         <p id="b11">
```

lorenzo6201, 2023-08-14:
can you post the entire file in a zip. I tried this code , but I think i did something wrong , because it does not work. I`m not that good with programing.

delboy711, 2023-08-14:
Sure.  I have applied it to my  fork of Stuart's repository at [https://github.com/delboy711/diyBMSv4ESP32](https://github.com/delboy711/diyBMSv4ESP32)
You can download a Zip file of the entire code there and compile it in the same way as you do Stuart's.
Are you OK with compiling in PlatformIO?

The affected files are 
ESPController/include/CurrentMonitorINA229.h
ESPController/src/CurrentMonitorINA229.cpp
ESPController/web_src/default.htm    (just trivial changes to the text. )

lorenzo6201, 2023-08-14:
i`m okay with compiling. Thank you