stuartpittaway/diyBMSv4ESP32 #215: module identify not working; controller reboots on button pressed

marcin11, 2023-06-09:
Modules, Identify button does nothing:
Compiled: 2023-04-17T08:23:56.977Z
Controller version 4 
Processor: ESP32
battery modules v 4.5

Same controller; when screen turns off for 15+ minutes (http server is working) and button/s on the PCB board are pressed the controller reboots

stuartpittaway, 2023-06-13:
Hi @marcin11 

What happens when you click on the identify button?  All that should occur is the relevant modules LED lights up for about 10 seconds (solidly).

Can I ask you to try the "charge_float" branch which is the next pre-release version?  

You can get the files from the bottom of this page... https://github.com/stuartpittaway/diyBMSv4ESP32/actions/runs/5256593144

lorenzo6201, 2023-06-14:
Module identify it does not work. i have the same issue.

stuartpittaway, 2023-06-14:
Ok, @lorenzo6201 you mentioned you have v4.5 of the cell modules - does the LED flash normally on those when processing a request? 

Are you running a recent version of the cell firmware/software on them?

Jestergnet, 2023-06-14:
I also do not have identification, when polled, the LEDs flicker. After pressing the ID button, nothing happens...

Board version 4.5
The latest firmware version.

lorenzo6201, 2023-06-16:
> Ok, @lorenzo6201 you mentioned you have v4.5 of the cell modules - does the LED flash normally on those when processing a request?
> 
> Are you running a recent version of the cell firmware/software on them?

Yes 4.5.
The led flashes normally when processing requests.
Latest firmware.
When pressing ID the green led flashes once shortly , after processing a  request is flashing one more time.

moscito1010, 2023-07-28:
I can confirm that for the 4.50 modules I use.
When I click on identify, the LED flashes exactly once. So every click lets the LED shine once. I can only remember the 4.21 modules where the LED flashed for about 10 seconds. The 4.40s flashed only briefly. I assumed until now that it was wanted.

stuartpittaway, 2023-07-29:
> LED flashed for about 10 seconds

This is what should happen with the identify - the LED stays on for about 10 seconds

closed 2023-09-08 by stuartpittaway