stuartpittaway/diyBMSv4ESP32 #216: PYLON CAN protocol to Deye Inverter not working for "charge prevented"

olterion, 2023-06-13:
**Describe the bug**
I have the Controller Board connected via CAN to my Inverter Deye SUN-12K- SG04LP3.
For the communication I have choosen the PYLON protocol in the diyBMS Controller.
My Inverter is recognizing the PYLON protocol and in normal operation mode everythin runs fine.
Normal operation means, the diyBMS has no injured rules or the charge steeings are set to chrarge at the moment.

But if the web Interface has an overlay window, which says "charging is prevented" my inverter has the warning "W31 Battery_comm_warn".
In this case the charge voltage and current are set to 0.1 from diyBMS and the "charge enabled" flag is not set.
In my Inverter I can read the 0.1V charge voltage, but the last current of 1.4A is even in the inverter.
Additionally it has this warning.

In total the inverter is still charging, although it shloudn't do.

Because I have no original PYLONTECH battery, I can't take a trace in the case, the bms want to prevent the charging.
Do you have such traces or do you have any idea why this behavior is still there?

If the charge preventing the complete system comes back to normal behavior, the warning in the inverter disappers and erverything runs fine.

**Hardware/Software Versions**
Controller version (from PCB): ESP32
(below can be obtained from the "About" page in the controller web interface)
Processor: ESP32
Version: [5b7135f8127c6fd9d5d18525f7d5de72a32b4232]
Compiled: 2023-04-17T08:23:56.977Z

**To Reproduce**
Steps to reproduce the behavior:
1. Set the charge/discharge settings
2. Wait until the inverter has charged the battery alomst complete
3. Take a look to the "charging is prevented" message
4. Look at the Deye inverter for Warning "W31 Battery_comm_warn"

stuartpittaway, 2023-06-13:
Hello, a number of inverters don't support the full PYLON communication protocols.

Can you send a screenshot of the charge settings page please?

olterion, 2023-06-13:
Hello Stuart,

thank you very much for your support.

Here is a screenshot of the charge/discharge settings:

[image]

stuartpittaway, 2023-06-14:
Is this a 16 cell setup?  

You have the cell maximum voltage (3400mV) but that is very much below the 56.5V total (3531mV/cell)  (56.5/16)

In normal operation, the "charge prevented" should be rarely seen.  I think you need to tweak the settings until you no longer get it.

These are the settings I'm running with, it generally charges my LIFEPO4 cells to 3450mV per cell - ignore the "Absorption timer (minutes)" that won't be in the version of code you are running at the moment (next release!)

[image]

olterion, 2023-06-14:
Hello,

yes I have a 16 cell LIFEPO4 Setup.
I will try your settings and will report it here.

olterion, 2023-06-15:
Hello Stuart,

i have setted your settings identical to my diyBMS controller.
But the complete setup runs still into the "charging is prevented" state.
I think, it is because the "Battery charge voltage (V)" is set to 56,0V and the actual voltage is over 56,0V?!

[image]

So in this case, the Warning is still present in the inverter.
In this state it isn't charging anymore, so the voltage drops to under 56,0V and the "charging is prevented" message disappears.
After that, the voltage goes again to over 56,0V and the game starts again.

Do you have any other ideas, why the inverter shows this warning?

Best regads

stuartpittaway, 2023-06-15:
The voltage shouldn't be exceeded by the inverter, it sounds like the inverter isn't regulating the voltage accurately.

Once the voltage goes over the settings, diybms will tell the inverter to stop (charge disabled). But this should clear once the voltage drops down.

stuartpittaway, 2023-06-23:
Thank you for taking the time to test and find a solution.  I'm sure we can incorporate this change into a future release.

stuartpittaway, 2023-08-09:
@olterion I've integrated the suggestions you have made into the next release of DIYBMS controller code.

Would you be able to test this for me?  The pre-compiled version of the code is here..

https://github.com/stuartpittaway/diyBMSv4ESP32/actions/runs/5807898981

olterion, 2023-08-15:
Hello Stuart,

I have uploaded the changes and it works very well.
I got the BMS into "charge prevented" mode and everything works as expected.

Thank you very much for implementing the change!

stuartpittaway, 2023-08-15:
Nice! Thanks for feedback

Adasbms, 2023-08-25:
Hello, can I write this version https://github.com/stuartpittaway/diyBMSv4ESP32/actions/runs/5807898981 to my DIYBMS because I also have a Deye Inverter.

If so, which one should be written?

[DIYBMS-Compiled]
(https://github.com/stuartpittaway/diyBMSv4ESP32/suites/14946646590/artifacts/852244889)
or
[DIYBMS-Release-Artifact-2023-08-09-10-34](https://github.com/stuartpittaway/diyBMSv4ESP32/suites/14946646590/artifacts/852244890)

Thank you very much

stuartpittaway, 2023-08-25:
Both files contain the same files, either is ok

stuartpittaway, 2023-10-30:
Hi @Adasbms did the latest firmware fix the DEYE issue for you?

Adasbms, 2023-10-30:
Hi, @stuartpittaway yes now the DEYE inverter works without problems.

[image]
[image]
[image]

Now I use All-in-one my configuration now is as follows.

Thank you very much

closed 2023-10-31 by stuartpittaway

stuartpittaway, 2023-11-27:
> NO Currentsensor !

This is a problem - you need a current sensor for any state of charge calculations - until then, the state of charge will be reported as zero and integration won't work properly.

Bingo2023, 2023-11-27:
> > NO Currentsensor !
> 
> This is a problem - you need a current sensor for any state of charge calculations - until then, the state of charge will be reported as zero and integration won't work properly.

ok Stuart, thanks for the comment.
That means the Victron implementation works with out a current sensor and the Pylontech implementation needs current sensor?

stuartpittaway, 2023-11-27:
Technically they both need it, but victron can chose to ignore.

Adasbms, 2023-11-27:
Hello, I checked the new version works with Deye inverter

Inverter:
SUN-12K-SG04LP3-EU
MAIN：2005-1140-1807

BMS:
latest Version ([Compiled_Firmware_2023-11-26-15-27.zip)
With current sensor 300A 50mv

[image]
[image]
[image]
[image]
[image]

Bingo2023, 2023-11-28:
@Adasbms: thank you very much for your details. That Proofs that it is possible.
I will compile a version with static values and analyse it.

I still can not belief that Deye is sending 0 for everything when 1 or 2 values are missing...
I would then expect correct voltages etc and an error/alarm or so.... let me find out.

stuartpittaway, 2023-11-29:
@Bingo2023 if you take a look at the controller code file ```pylon_canbus.cpp```, there are a lot of sections which don't run when a current monitor is missing - generally wrapped in conditions like this...
```if (mysettings.currentMonitoringEnabled && currentMonitor.validReadings)```

Just a reminder that its critical to have current monitoring and state of charge calculations for LIFEPO4 cells - if you are using a different chemistry, then its not as bad as you can use the terminal charge voltage as an indicator (but DIYBMS doesn't do this without the current monitor/shunt).

Bingo2023, 2023-11-29:
wow by adding the message 0x355 (SOC and SOH) the Deye shows values send via CAN....