stuartpittaway/diyBMSv4ESP32 #217: bms command to Solis inverter issue

anthonyevs, 2023-06-14:
**Describe the bug**
A clear and concise description of what the bug is.

**Hardware/Software Versions**
Controller version (from PCB): ??
(below can be obtained from the "About" page in the controller web interface)
Processor: ??
Version: ???
Compiled: ??

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

stuartpittaway, 2023-06-14:
Did you miss off the problem/description?

anthonyevs, 2023-06-14:
Hi Stuart,
i have the controller communicating to the inverter, it reports voltage an SOC correctly.
Onve the inverter receives a command from the BMS to stop discharge due to low SOC the inverter then just stops, it doesnt start a charge sequence when there is solar energy availabe, and displays a msg StandbySynoch. Im thinking maybe the inverter isnt getting a command to restart a charge after a dischsrge.
Any thoughts.

stuartpittaway, 2023-06-14:
Hi @anthonyevs 

The BMS doesn't actually send a message to stop charging unless something went wrong.

All it does is send the voltage and current requirements to the inverter (these values are shown in the "Dyn. Chg." tiles.)

Do you know if there are any datasheets or communication protocol documents for your inverter?

Stu.

stuartpittaway, 2023-06-23:
Hi @anthonyevs is this still happening?

anthonyevs, 2023-06-29:
yes stuart, im hoping with fine tuning of the bms it will stop the issue, or, i can try to select "owner defined battery on the inverter and add in my own battery settings.
A Solis engineer told me that Solis now refuse to accept any support for owners with their own batteries, so i`m not going to ask them for canbus data as they will then know im running my own battery and may refuse warranty.
Thanks

stuartpittaway, 2023-06-29:
Ok, there isn't anything obvious I can do or change from my side, I'll leave the issue open and we can review as the project develops.

stuartpittaway, 2023-11-28:
Hi @anthonyevs with the latest controller firmware, do you still have this problem?

anthonyevs, 2023-11-28:
Hi Stuart, its all working now, initially with the fix you provided for the 
deye inverter and now with the latest revision its all working fine. Thank you for your support, and continued support, its a great project and developing all the time, as with the newly relesed version and now home assistant compatible. Brilliant. Waiting now for the availabilty wityh yourself to supply the newer all in 1 boards for our next projects. Thanks.

closed 2023-11-28 by stuartpittaway