stuartpittaway/diyBMSv4ESP32 #218: Controller rebooting

lorenzo6201, 2023-06-14:
**Describe the bug**
When enabling Influx DB integration the controller reboots. The seconds between reboots is equal to the logging frequency.

**Hardware/Software Versions**
Controller version (from PCB): 7a77390
(below can be obtained from the "About" page in the controller web interface)
Processor:  ESP32
Version: [5b7135f8127c6fd9d5d18525f7d5de72a32b4232]
Compiled: 2023-04-17T08:23:56.977Z

**To Reproduce**
Steps to reproduce the behavior:
1. Go to integrations and activate influx DB
2. Click on save

The controller reboots only when inna229 addon board is fitted. without the board , the controller works fine
The rebooting happens if powered by USB or by Buck converter.

USB logs below
`
               _          __
  _|  o       |_)  |\/|  (_
 (_|  |  \/   |_)  |  |  __)
         /

CONTROLLER - ver:5b7135f8127c6fd9d5d18525f7d5de72a32b4232 compiled 2023-04-17T08:23:56.977Z
ESP32 Chip model = 1, Rev 3, Cores=2, Features=50
I (70) diybms-hal: Configure I2C
I (73) diybms-hal: Scanning i2c bus
I (79) diybms-hal: Found i2c device at address 0x21
I (89) diybms-hal: Found TCA6416A
D (91) diybms-hal: Configure VSPI
I (92) diybms: ** Controller changed state from Unknown to PowerUp **
I (94) diybms-set: nvs_flash_init
I (96) diybms-hal: CAN driver installed.  Filter=1621098496 Mask=6291455
I (101) diybms-hal: CAN driver started
I (111) diybms: LittleFS mounted, total=524288, used=139264
I (202) diybms: Mounting SD card
I (202) diybms-hal: Mounting SD card
[   764][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[   764][W][sd_diskio.cpp:512] ff_sd_initialize(): sdWait fail ignored, card initialize continues
[   766][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[   773][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  1281][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  1281][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
E (1221) diybms-hal: Card mount failed

Press SPACE BAR to enter console WiFi configuration....

....................

No key press detected
I (6230) diybms-set: Load WIFI config
I (6231) diybms-set: Read key (SSID)
I (6232) diybms-set: Read key (PASS)
I (6232) diybms-set: Load config
I (6235) diybms-set: Count: UsedEntries = (230), FreeEntries = (400), AllEntries = (630)
I (6243) diybms-set: Read key (totalBanks)
I (6246) diybms-set: Read key (totalSeriesMod)
I (6250) diybms-set: Read key (baudRate)
I (6254) diybms-set: Read key (interpacketgap)
I (6259) diybms-set: Read key (rulevalue)
I (6263) diybms-set: Read key (rulehysteresis)
I (6267) diybms-set: Read key (rulerelaystate)
I (6271) diybms-set: Read key (rulerelaydef)
I (6275) diybms-set: Read key (relaytype)
I (6278) diybms-set: Read key (g_voltagehigh)
I (6282) diybms-set: Read key (g_voltagelow)
I (6286) diybms-set: Read key (BypassOverTemp)
I (6290) diybms-set: Read key (BypassThresmV)
[... 148 lines cut ...]
ESP32 Chip model = 1, Rev 3, Cores=2, Features=50
I (70) diybms-hal: Configure I2C
I (73) diybms-hal: Scanning i2c bus
I (79) diybms-hal: Found i2c device at address 0x21
I (89) diybms-hal: Found TCA6416A
D (91) diybms-hal: Configure VSPI
I (92) diybms: ** Controller changed state from Unknown to PowerUp **
I (94) diybms-set: nvs_flash_init
I (96) diybms-hal: CAN driver installed.  Filter=1621098496 Mask=6291455
I (101) diybms-hal: CAN driver started
I (111) diybms: LittleFS mounted, total=524288, used=139264
I (202) diybms: Mounting SD card
I (202) diybms-hal: Mounting SD card
[   764][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[   764][W][sd_diskio.cpp:512] ff_sd_initialize(): sdWait fail ignored, card initialize continues
[   766][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[   773][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  1281][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  1281][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
E (1221) diybms-hal: Card mount failed

Press SPACE BAR to enter console WiFi configuration....

.................
`

stuartpittaway, 2023-06-14:
@lorenzo6201 looking at the code, the message you see in the logs should never appear!

When this error happens, are your cell modules connected?  Do the bar graphs appear in the web page?

```
    // If we did not generate any module data we can exit early, although this should never happen.
    if (module_data.empty() || module_data.length() == 0)
    {
        ESP_LOGI(TAG, "No module data to send to InfluxDB");
        return;
    }
```

atanisoft, 2023-06-14:
@stuartpittaway the reboot looks likely to be either a mutex is being taken but not returned or attempting to take while already taken. Double check that you are not missing a check to release the mutex somewhere.

stuartpittaway, 2023-06-14:
Agreed @atanisoft, backtrace 

```
xtensa-esp32-elf-addr2line.exe -fe C:\temp\diybms_controller_firmware_espressif32_esp32-devkitc.elf 0x40084189:0x3ffce070 0x4008e4fd:0x3ffce090 0x40095299:0x3ffce0b0 0x4008f50d:0x3ffce1e0 0x400e9108:0x3ffce220 0x400d3bd4:0x3ffce250 0x400d3ce4:0x3ffce270 0x400d3d19:0x3ffce290 0x400d3f42:0x3ffce2b0 0x400d988f:0x3ffce300
```

```
panic_abort
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408     
esp_system_abort
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
__assert_func
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
xQueueSemaphoreTake
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c:1549 (discriminator 1)
_ZN8SPIClass16beginTransactionE11SPISettings
_ZN20CurrentMonitorINA22914spi_readUint24ENS_12INA_REGISTERE
/home/runner/work/diyBMSv4ESP32/diyBMSv4ESP32/ESPController/src/CurrentMonitorINA229.cpp:138
_ZN20CurrentMonitorINA22910readUInt24ENS_12INA_REGISTERE
/home/runner/work/diyBMSv4ESP32/diyBMSv4ESP32/ESPController/src/CurrentMonitorINA229.cpp:207
_ZN20CurrentMonitorINA22910BusVoltageEv
/home/runner/work/diyBMSv4ESP32/diyBMSv4ESP32/ESPController/src/CurrentMonitorINA229.cpp:230
_ZN20CurrentMonitorINA22912TakeReadingsEv
/home/runner/work/diyBMSv4ESP32/diyBMSv4ESP32/ESPController/src/CurrentMonitorINA229.cpp:275
_Z8rs485_txPv
/home/runner/work/diyBMSv4ESP32/diyBMSv4ESP32/ESPController/src/main.cpp:2935
```

Which seems to relate to (main.cpp)
```
// Take readings from internal INA229 chip (on controller board)
          if (hal.GetVSPIMutex())
          {
            currentmon_internal.TakeReadings();
            ProcessDIYBMSCurrentMonitorInternal();
            hal.ReleaseVSPIMutex();
          }
```

The crash occurs in part of the ```TakeReadings()``` code, when it makes a SPI call to the INA229 chip.

However, what's confusing, is the inference that enabling INFLUX triggers this error - its got nothing to do with this code.

@lorenzo6201 is that correct?  Does this error occur if INFLUX is disabled?

lorenzo6201, 2023-06-14:
if the influx is disabled , the controller works like a charm. yes the cell modules were connected to the controller

lorenzo6201, 2023-06-14:
I tried the new firmware you released on patreon. The influx db issue is fixed and the database is populated, controller no longer reboots , the cell identify still does not work.

stuartpittaway, 2023-06-14:
Which is even more strange as there are no changes to either influx or current monitoring in that release!

I'm starting to think I keep running in to issues with the esp32 silicon and bugs in that, or the IDF framework

lorenzo6201, 2023-06-18:
Found another bug. If Mqtt is enabled , every 20-30 seconds the controller gives an error that it can`t communicate with modules and/or shunt. If mqtt is disabled no issues. The same applies to the next release. 
I had the internal bms error rule active to disconnect the battery when triggered and notice the banks started to disconnect one by one. took me all day today trying to figure it out. I changed the cables between controller and modules with two wire twisted shielded cable , I fitted a ferrite core on each pair of cables. The cables from controller to modules are 40 mm long . Finally I flashed the firmware again on a controller  and the issue went away. Tested the bank for one hour and seemed ok , activated the mqtt and once again the issue appeared. Disabled mqtt on all controllers and  no issues for over 2 hours.

stuartpittaway, 2023-06-19:
@lorenzo6201 I've opened another issue for the MQTT  however - I've been running this software for 9months with MQTT running perfectly and I don't see any controller issues being reported.  - can we pick up this conversation over at #219

stuartpittaway, 2023-06-19:
@lorenzo6201 are you using the pre-compiled firmware from GITHUB or compiling your own code?

stuartpittaway, 2023-06-23:
@lorenzo6201 I'm struggling to find anything obviously wrong, but given that you have issues with Influx and MQTT - perhaps theres a problem with the WIFI/network signal ?

Its going to be very difficult to locate a problem without the serial USB logs from the ESP32 - if you can capture those, it may highlight an error.

lorenzo6201, 2023-06-23:
i doubt is the wifi signal. All controllers are connected to a separate router that is literally  30 cm away. I will try to capture the serial usb log. I have one controller on the bench right now with menningd firmware on it. I will reflash it an trying to replicate the issue.

stuartpittaway, 2023-07-08:
Going to close this issue for INFLUX as the MQTT one replaces it

closed 2023-07-08 by stuartpittaway