stuartpittaway/diyBMSv4ESP32 #221: diyBMSv4 w/ onboard current monitor got kernel panic after writing on SD card

mgodziszewski, 2023-06-23:
The problem appears on the diyBMSv4 ESP32 controller with an integrated current monitor working under Compiled_Firmware_2023-04-17-08-26 when, for example, saving the Wi-Fi configuration to the SD card:

Here you can find the logs from the console when successfully writing data on the same physical controller, but working with Compiled_Firmware_2023-03-29-08-24

I (46544) diybms: Task 1
I (49017) diybms-webpost: API post: wificonfigtofile
**I (49017) diybms-webpost: Creating folder
I (49019) diybms-webpost: Generating SD file /diybms/wifi.json
I (49027) diybms-webpost: Delete existing file /diybms/wifi.json**
D (49079) diybms: Send addr=90, func=3, len=8
I (49110) diybms-rules: Set warning 1

After updating the firmware to Compiled_Firmware_2023-04-17-08-26 here you can find logs for the same operation (saving wifi configuration on SD card), unfortunately ended with a kernel panic:

**[ 67344][W][sd_diskio.cpp:186] sdCommand(): token error [13] 0x11
[ 67345][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
[ 67346][W][sd_diskio.cpp:180] sdCommand(): crc error**
I (75409) diybms: Task 1
D (75499) diybms: Open log /data_20230623.csv
D (75539) diybms: Wrote to SD log
D (75564) diybms: Wrote current monitor data to SD log
E (75686) diybms: Fail to queue CANBUS message (0x107)
I (75859) diybms-rules: Set warning 1
E (75936) diybms: Fail to queue CANBUS message (0x107)
E (76286) diybms: Fail to queue CANBUS message (0x107)
E (76536) diybms: Fail to queue CANBUS message (0x107)
E (76786) diybms: Fail to queue CANBUS message (0x107)
E (77037) diybms: Fail to queue CANBUS message (0x107)
E (77287) diybms: Fail to queue CANBUS message (0x107)
I (77535) diybms-webreq: API call: monitor2
I (78025) diybms-webreq: API call: monitor2
E (78537) diybms: Fail to queue CANBUS message (0x107)
I (78861) diybms-rules: Set warning 1
E (78887) diybms: Fail to queue CANBUS message (0x107)
E (79137) diybms: Fail to queue CANBUS message (0x107)
E (79387) diybms: Fail to queue CANBUS message (0x107)
E (79637) diybms: Fail to queue CANBUS message (0x107)
E (79887) diybms: Fail to queue CANBUS message (0x107)
**I (80054) diybms-webpost: API post: wificonfigtofile
I (80055) diybms-webpost: Creating folder
[ 72033][W][sd_diskio.cpp:186] sdCommand(): token error [13] 0x11
[ 72035][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed
[ 72041][W][sd_diskio.cpp:180] sdCommand(): crc error
E (80137) diybms: Fail to queue CANBUS message (0x107)
I (80177) diybms-webpost: Generating SD file /diybms/wifi.json
I (80183) diybms-webpost: Delete existing file /diybms/wifi.json**
E (80487) diybms: Fail to queue CANBUS message (0x107)
E (80737) diybms: Fail to queue CANBUS message (0x107)
I (80910) diybms: Task 2
E (81087) diybms: Fail to queue CANBUS message (0x107)
I (81099) diybms-webreq: API call: monitor2
E (81338) diybms: Fail to queue CANBUS message (0x107)
E (81589) diybms: Fail to queue CANBUS message (0x107)
I (81598) diybms-webreq: API call: monitor2
E (81839) diybms: Fail to queue CANBUS message (0x107)
I (81863) diybms-rules: Set warning 1
E (82090) diybms: Fail to queue CANBUS message (0x107)
E (83340) diybms: Fail to queue CANBUS message (0x107)
E (83690) diybms: Fail to queue CANBUS message (0x107)
E (83940) diybms: Fail to queue CANBUS message (0x107)
E (84190) diybms: Fail to queue CANBUS message (0x107)
E (84440) diybms: Fail to queue CANBUS message (0x107)
I (84677) diybms-webreq: API call: monitor2
E (84690) diybms: Fail to queue CANBUS message (0x107)
I (84864) diybms-rules: Set warning 1
E (84889) diybms: CANBUS timeout
E (84942) diybms: Fail to queue CANBUS message (0x107)
I (85172) diybms-webreq: API call: monitor2
E (85292) diybms: Fail to queue CANBUS message (0x107)
E (85542) diybms: Fail to queue CANBUS message (0x107)
E (85892) diybms: Fail to queue CANBUS message (0x107)
E (86142) diybms: Fail to queue CANBUS message (0x107)
E (86393) diybms: Fail to queue CANBUS message (0x107)
D (86411) diybms: Task 3, s=0 e=15
E (86643) diybms: Fail to queue CANBUS message (0x107)
E (86893) diybms: Fail to queue CANBUS message (0x107)
I (87866) diybms-rules: Set warning 1
E (88143) diybms: Fail to queue CANBUS message (0x107)
I (88241) diybms-webreq: API call: monitor2
E (88493) diybms: Fail to queue CANBUS message (0x107)
I (88741) diybms-webreq: API call: monitor2
E (88743) diybms: Fail to queue CANBUS message (0x107)
E (88998) diybms: Fail to queue CANBUS message (0x107)
E (89248) diybms: Fail to queue CANBUS message (0x107)
E (89498) diybms: Fail to queue CANBUS message (0x107)
E (89748) diybms: Fail to queue CANBUS message (0x107)
**Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40093e8f  PS      : 0x00060f33  A0      : 0x80094d64  A1      : 0x3ffb2190
A2      : 0x8006c4ac  A3      : 0x40094d9c  A4      : 0x3ffb21f4  A5      : 0x3ffe7184
A6      : 0xfffffffc  A7      : 0x00000001  A8      : 0x80093ea9  A9      : 0x3ffb2170
A10     : 0x3ffe7184  A11     : 0x4008532c  A12     : 0x00000001  A13     : 0x3ffb21f4
A14     : 0x00060f20  A15     : 0x00000001  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c
EXCVADDR: 0x8006c4b0  LBEG    : 0x4008a2b4  LEND    : 0x4008a2bf  LCOUNT  : 0x00000000

Backtrace: 0x40093e8c:0x3ffb2190 0x40094d61:0x3ffb21b0 0x40094fb3:0x3ffb21d0 0x4011e3aa:0x3ffb21f0 0x400da091:0x3ffb2230 0x400f1e11:0x3ffb2290

ELF file SHA256: 837dfffd9b1850d8

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
[    60][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
I (56) diybms:

               _          __
  _|  o       |_)  |\/|  (_
 (_|  |  \/   |_)  |  |  __)
         /

CONTROLLER - ver:5b7135f8127c6fd9d5d18525f7d5de72a32b4232 compiled 2023-04-17T08:23:56.977Z
ESP32 Chip model = 1, Rev 1, Cores=2, Features=50
I (71) diybms-hal: Configure I2C
I (73) diybms-hal: Scanning i2c bus
I (79) diybms-hal: Found i2c device at address 0x21
I (90) diybms-hal: Found TCA6416A
D (91) diybms-hal: Configure VSPI
I (93) diybms: ** Controller changed state from Unknown to PowerUp **
I (452) diybms-set: nvs_flash_init
I (454) diybms-hal: CAN driver installed.  Filter=1621098496 Mask=6291455
I (454) diybms-hal: CAN driver started
I (467) diybms: LittleFS mounted, total=524288, used=192512
I (706) diybms: Mounting SD card
I (706) diybms-hal: Mounting SD card
I (716) diybms-hal: SD card available**

Press SPACE BAR to enter console WiFi configuration....

mgodziszewski, 2023-06-23:
This happens every time during a write operation to the SD card

stuartpittaway, 2023-06-23:
Thanks for reporting this. A number of people have issues with the SD card at the moment, I believe it may be related to the underlying Arduino library but as yet unproven.

Are you using the microsd card slot on the board or the SD card slot on the TFT display?

How are you powering the controller PCB?

mgodziszewski, 2023-06-23:
Stuart,
I'm using SD card onboard. PCB is powered from cells via dc2dc (48V/5V) converter.

Kind regards,
Michal

stuartpittaway, 2023-07-08:
Thanks Michal.

Can I ask you to try the next version of the controller code - its available in this file.

https://github.com/stuartpittaway/diyBMSv4ESP32/suites/14152731462/artifacts/792907330

Let me know if you still see errors.

What size SD card are you using?

mgodziszewski, 2023-07-15:
Hi Stuart,
I can confirm that the 32G micro sd card is now active and mounted.

moscito1010, 2023-07-28:
> Are you using the microsd card slot on the board or the SD card slot on the TFT display?

Sorry but I just have to ask :-)
Is it possible to use the SD card slot on the display with it? Furthermore, I've been wondering for a long time whether the 4 lines from the display should be connected to the board.

stuartpittaway, 2023-07-29:
It should be possible to use the SD card slot on the display (it needs those 4 pins you mention @moscito1010) I have seen some issues with some cards - might be a voltage and/or current issue with certain brands of cards.

moscito1010, 2023-07-29:
SANDISK ULTRA 32GB does not seem to work. I have a whole collection of these for my camera and none of them can be used. I have some old 8GB from the same manufacturer, they don't work either. The controller apparently ate an 8G and a 32GB noname. After that I couldn't use it anymore. However, they were quite old, it's possible that they just needed a nudge to cross the Jordan :-) A cheap noname 32GB SD from ALDI, on the other hand, works great.

Maybe using the slot in the display is a solution and the cards work more reliably there, or is that technically impossible?

stuartpittaway, 2023-07-29:
> Maybe using the slot in the display is a solution and the cards work more reliably there, or is that technically impossible?

The SD card slot in the TFT screen does work (just don't use both at the same time)

stuartpittaway, 2023-08-02:
@mgodziszewski is this issue now resolved?  Can I close this?

mgodziszewski, 2023-08-02:
Stuart,Yes, I Can confirm that the latest fix solved as card problem.