stuartpittaway/diyBMSv4ESP32 #222: Not a bug, but sort of rquest

anthonyevs, 2023-06-27:
Been testing the latest code for the controller with new charge settings incorporated and few other things.
Possible request.
We have battery storage and anyone with an EV car charger whether its "smart" or not will realise that all our smart devices are smart on their own but they dont communicate to each other to be really smart.
So, that said, in the new firmware there is an option to prevent discharge.
Can a timer be added to this function to say prevent discharge between 0030 and 0430 (octopus cheap rate)
I ask as this is when the EV gets is charge, and i dont want to discharge my battery to charge the EV, id much prefer to let the EV charge under the cheap rate, plus the minimum house load current at this time is very low anyway.
This way i keep the battery for my dailyhouse loads at rates that are most expensive.
Hope this makes kind of sense.
Thanks Stu

stuartpittaway, 2023-06-28:
Hi @anthonyevs normally this type of functionality would be found in the inverter/charger.  My inverter/charger has a time of use mode for this exact purpose.

Does yours have anything similar?

I'm finding that the "prevent discharge" isn't as reliable with some inverters ignoring that request from the BMS.

anthonyevs, 2023-06-28:
Hi Stuart, sadly my inverter doesnt have the time slots in the menu.
Your BMS does tho on my inverter accept the stop discharge command very well, it works great.
Someday someone will devise a device to mane all our smart things work as a team together, as currently they are indiviually smart but still dumb in a working environment with other smart things.
Thanks
Anthony.

stuartpittaway, 2023-06-28:
Which inverter make/model do you have?