stuartpittaway/diyBMSv4ESP32 #223: Latest esp32 firmware (float style menu option) current shunt now not recording current accurately.

anthonyevs, 2023-06-28:
Hi Stuart,
Downloaed the latest code from patreon for the esp32 with the absorb float style menu options.
What i found is my current shunt (external board) now reads incorrectly, the bms reads the current very low, when working the current draw from the battery was 3.2kw, but the shunt was recording a reading of 2.2kw.
It was reading very accurately on the previous version firmware.
Is there anything i can do my end in settings to rectify the reading error.
Thanks.
Anthony.

stuartpittaway, 2023-06-28:
Ok, first of all, double check the settings for the shunt are correct on the web interface.

If you upgraded from a very old version, the settings may be misaligned.

Secondly, when was the last time you updated the code on the shunt?  That also changed earlier this year, so make sure you are running the latest code on that.

anthonyevs, 2023-06-29:
Hi Stuart, this is the print screenshot of the current shunt settings.
I have a new version of the shunt firmware, loaded about 8 weeks ago.
Let me know if you see anyhting obvious in the settings.
[image]

stuartpittaway, 2023-06-29:
That all looks fine - a 100amp shunt is quite small if you are pulling 3kw+ through it - normally, shunts should be at least 25% larger than needed - but it shouldn't change the measurements between firmware versions.

If you go back to an older firmware version on the controller, does the problem go away?

Nobody has reported this issue, and I would have expected a lot of complaints!

anthonyevs, 2023-06-29:
Oops, basic error on my part. I entered 100A in the current shunt setting thinking that is the max output of my inverter.
The shunt is a 150A shunt.
Changed to 150A shunt in the menu, now all is good.
Sorry for this, my bad.
Thanks.

closed 2023-07-08 by stuartpittaway