stuartpittaway/diyBMSv4ESP32 #227: Does anyone know the CANBUS pin assignment of a Voltronic Axpert MAX or MPPSOLAR or identical/similar devices?

moscito1010, 2023-07-28:
Does anyone know the CANBUS pin assignment of a Voltronic Axpert MAX or MPPSOLAR or identical/similar devices?
I want to connect the DIY controller to the Axpert MAX and the information I can find is conflicting.

moscito1010, 2023-07-29:
I've been experimenting a bit because I'm building a new pack right now. I haven't finished building all the cell modules yet, but it should be enough for a test.

According to this source, 6 is CAN-H, 7 is CAN-L, 8 is GND
[LINK](https://powerforum.co.za/uploads/monthly_2022_10/4191291F-06BF-467A-A918-7472B3E9C387.jpeg.17b6aeed1067303910e7f6ff46551fec.jpeg)

On the DIY controller, J7 has 1 +5V, 2 GND, 3 CAN-H, 4 CAN-L
+5v is not needed so would be RJ45 <-> DIYBMS
6<->3
7<->4
8<->3
If the source is correct, the cable should work. FYI this is a CAT5 network cable that I cut off at one end.

Here is a picture of my settings in the BMS.
[image]

[image]

With the inverter, I switched on the support for PYL (Pylontech) in menu item 5 and checked it several times.

stuartpittaway, 2023-08-02:
@moscito1010 did you get the inverter working over CANBUS ?

Jestergnet, 2023-08-02:
I was also looking for information about working with CAN BUS, but everywhere I came across that Axpert Max and similar millets do not have support, only RS485.

I'm just saying what I found on the net, maybe someone has other information?

stuartpittaway, 2023-08-02:
There is some experimental work on RS485 communication with inverters going on over here - https://github.com/stuartpittaway/diyBMSv4ESP32/pull/195
The development is by a another user of DIYBMS.

moscito1010, 2023-08-02:
> I was also looking for information about working with CAN BUS, but everywhere I came across that Axpert Max and similar millets do not have support, only RS485.
> 
> I'm just saying what I found on the net, maybe someone has other information?

sorry i found the discription in german
     Anpassbare LED-Statusleiste mit RGB-Leuchten
    Eingebautes Wi-Fi für die mobile Überwachung (Android / iOS-Apps sind verfügbar)
    Unterstützt die USB On-the-Go-Funktion
    Kommunikationsanschluss für BMS reserviert (RS485, CAN-BUS oder RS232)
    Austauschbares Lüfterdesign für einfache Wartung
    Batterieunabhängiges Design
    Konfigurierbare Priorisierung und Timer der AC / PV-Ausgangsnutzung
    Wählbarer Hochleistungsladestrom
    Wählbarer Eingangsspannungsbereich für Haushaltsgeräte und PCs
    Kompatibel mit Netz- oder Generatoreingang
    Anti-Staub-Kit
    Optionaler Gleichstromausgang für Gleichstromlüfter, LED-Lampe, Router usw.
    Parallelbetrieb bis zu 6 Einheiten

moscito1010, 2023-08-02:
OK, I'm just testing the pylontech_rs485 branch.

chapulino, 2023-08-27:
finally someone has been able to connect a voltronic type inverter by cambus to the DIYBMS with pylontech emulation? What were the communication pins between one and the other?

moscito1010, 2023-08-28:
No, not CAN but RS485 (Modbus). I have Axpert Max 7.2 inverter and had also tried it with CAN but actually the CANBUS does not seem to work with the inverter and it communicates with the Pylontech batteries via MODBUS.
It works fine with the RS485 branch.

Here is the pinout I used
Inverter<->BMS (RS485)
5<->A
3<->B
I installed this branch
https://github.com/ruza87/diyBMSv4ESP32/tree/pylon_rs485

chapulino, 2023-08-28:
Thanks, do you know if it will be compatible with the All in One?
Could an update be made to work with Aspert but to collect the updates from the latest controller?

moscito1010, 2023-08-28:
You mean Stuart's all-in-one hardware?
I don't know if the RS485 firmware is compatible with this hardware. If I understood correctly, this hardware is still experimental and therefore not relevant for me. I don't see any real advantage in the ALL-IN-ONE board compared to the individual modules, apart from better cable management. Somehow I imagined it differently, but that's another topic and doesn't belong here.

chapulino, 2023-08-28:
thanks.
Last question from which controller port do you connect the Controller to the inverter from RS485 or from CAMBUS?

moscito1010, 2023-08-28:
I am use the the Port with the battery pictogram

chapulino, 2023-08-29:
Thanks, but the question was about the DIBMS controller port. The one marked RS485 or the one labeled CAMBUS

[image]

moscito1010, 2023-08-29:
I Have the same inverter, it is the battery Port. This is the left rj45
Port. In 30min I am at home and i can send any pictures

chapulino, 2023-08-29:
the doubt is not with the inverter.

The question is how the inverter is connected to the DIYBMS controller from which port: RS485 or CAMBUS of the DIYBMS controller. The data is sent to the inverter.
Sorry for the inconveniences

moscito1010, 2023-08-29:
[image]
[image]
I connect with the left RJ45 Port from the Inverter to RS485 on the BMS Controller

The Voltronic Axpert MAX series and others with this display as well as the successors with the large round display cannot use a CAN protocol, even if it says so in the description.
I borrowed an original Pylontech to Voltronic cable and measured it. The Pylontech batteries can use MODBUS and CANBUS but MODBUS is used for the Voltronic devices.
I really tested every possibility to connect the Axpert Max devices to the DIYBMS via CANBUS. That was an unsuccessful afternoon :-)
Since I have 3 Axpert MAX from 3 OEMs, I wrote to every support (MPPsolar, Voltacon UK, SP24 Italy) Neither the OEMs nor Voltronic themselves give you a useful answer.

chapulino, 2023-08-29:
Thank you, tonight we will tell you if we succeed, we are going to make a
bus from the rs485 port of the controller to the inverter and to the
current controller (both are connected to the port, as the Shunt has a
terminating resistor, it will be the last on the bus, and We will test
whether the JP4 of the controller should be open or closed and we will tell
you and thank you very much for the explanations

stuartpittaway, 2023-08-29:
@chapulino you may run into issues on the RS485 if both shunt and inverter are connected - do they run at the same baud rate/parity ?

moscito1010, 2023-08-29:
Maybe IT is possible for you. I am use the little shunt addon and not the
smart shunt.

chapulino, 2023-08-29:
> @chapulinoEs posible que tenga problemas en el RS485 si tanto la derivación como el inversor están conectados. ¿Funcionan a la misma velocidad/paridad de baudios?

we do not know , this afternoon we will try all the options , with and without Current Shunt
You can verify that if it controls the derivation of the all in one, although logically without the balancing functions by deviation of mean value and the activation of the heatsink fan, the all in one must be previously programmed on a controller with the latest firmware version so that this fan activation parameter is saved.

moscito1010, 2023-08-29:
Sorry...what a Fan? A Fan for cooling the Shunt ?

chapulino, 2023-08-29:
> Sorry...what a Fan? A Fan for cooling the Shunt ?

no, I mean the fan of the heatsink of the All in One resistors, but for now we focus on the RS485 communication

chapulino, 2023-08-31:
> I found some information in the pylontech documentation that might be
> helpful.
> [image: grafik]
> <https://user-images.githubusercontent.com/75945899/264462794-38c9f88e-613c-48be-8424-d5fac886bff2.png>
>
> https://portal.segensolar.de/reseller/docs/Pylontech%20Compatible%20List%20V2.15%20ESS.pdf

Thank you for your interest, the truth is that we have tried all the
possible options but if it has been successful so far, at one point we even
burned the ICs that provide RS485 service, it has already been replaced,

We have several questions that may help us.

Do you currently have the inverter in service with the DIYBMS?
Do you have the DIYBMS Curren controller connected?
Is it the internal or external model?
If it is external, is the DIYBMS connection to the inverter through an
RS485 BUS?
We have seen that the DIYBMS Current Contoler can have its BaudRate
modified so that it has the same as the 9600 inverter.
Do you have a 120ohm resistor between A and B on the inverter side?
Do you have to activate RS485 communication or is it activated in your
firmware by default?
In the tests carried out, it seems that if the Current controller via
Modbus is not activated, RS485 communication is not enabled.

I'm sorry, we are very lost and we can't get the firm to work.

moscito1010, 2023-09-02:
I use 3 Axpert MAX 7.2KW inverters with a DIYBMS and the RS485 firmware
I installed this branch
https://github.com/ruza87/diyBMSv4ESP32/tree/pylon_rs485
I use the Current ADDON not the external shunt.
I use RS485 to connect to the AXPERT MAX.
I set the solder bridge for RS485 (solder on the two contacts, no extra resistance)
Only H + L must be connected to the RS485 port, voltage and ground have no place there.
The PLY protocol must be activated on the inverter.
[image]
[image]

Jestergnet, 2023-09-02:
> I use 3 Axpert MAX 7.2KW inverters with a DIYBMS and the RS485 firmware I installed this branch https://github.com/ruza87/diyBMSv4ESP32/tree/pylon_rs485 I use the Current ADDON not the external [...]
Hello, what is the pattern of RJ45 on the inverter side?

moscito1010, 2023-09-02:
Colleague, do you also read what we write? :-)
https://github.com/stuartpittaway/diyBMSv4ESP32/issues/227#issuecomment-1695173258

Jestergnet, 2023-09-02:
> Colleague, do you also read what we write? :-)

I'm sorry. I saw above the message, but did not attach importance.

moscito1010, 2023-09-02:
> > Sorry...what a Fan? A Fan for cooling the Shunt ?
> 
> no, I mean the fan of the heatsink of the All in One resistors, but for now we focus on the RS485 communication

sorry i don´t no what you mean.

Jestergnet, 2023-09-02:
> > > Sorry...what a Fan? A Fan for cooling the Shunt ?
> > 
> > 
> > no, I mean the fan of the heatsink of the All in One resistors, but for now we focus on the RS485 communication
> 
> sorry i don´t no what you mean.

I read the message above about the pinout, but did not betray the importance.

moscito1010, 2023-09-02:
> > Colleague, do you also read what we write? :-)
> 
> I'm sorry. I saw above the message, but did not attach importance.
[image]

chapulino, 2023-09-02:
Thank you very much for the explanation, the truth is that in the end my friend, after many tests, was able to establish communication between the DIYBMS BMS and his Axpert inverter (Voltronic variant)
It has a 15-module 18650 LITHIUM battery, so the voltages are a little higher than those of Pylontech, and the inverter does not allow these parameters to be adjusted. If a Pylontech is chosen as the battery, the voltage parameters, etc. remain the same. fixed...

He is going to modify the battery to 14 cells to continue testing. We will report the progress here, I reiterate, thank you very much.

Fernank2, 2023-09-10:
Hello,
I have a 15 Series 21700 cell powerwall with an Axpert IV MKS inverter and the DiyBms with internal current sensor.

I have installed the DiyBms version of Ruza87 with support for Pylontech RS485 batteries.

When I connect the battery and communication is established between DiyBms --> inverter (approximately 10 seconds) the inverter shows me error 60.

The axpert manual says the following about this error 60:

"If battery status is not allowed to charge and discharge after the communication
between the inverter and battery is successful, it will show code 60 to stop charging and
discharging battery."

What state is this?

On the other hand, some of the values ​​in the DiyBMS load configuration are written to the inverter.
parameter

max. chargin current param. 2
bulk chargin voltage param. 26
float chargin voltage param. 27
cutting low voltage param. 29

Should it write DiyBMS in the inverter?
Can you help me with this error 60?

moscito1010, 2023-09-13:
yes, it should write to the charge controller.
60 means no connection to the BMS.
the RS495 cable is 2 wires. You don't need GND or voltage, just the two data lines.

Fernank2, 2023-09-13:
> yes, it should write to the charge controller.
> 60 means no connection to the BMS.
> the RS495 cable is 2 wires. You don't need GND or voltage, just the two data lines.

The communication failure with the BMS is error 61. I have error 60. The BMS communicates with the inverter because the BMS writes the parameters to the inverter.

Fernank2, 2023-09-14:
> […](#)

Yes

moscito1010, 2023-09-14:
@Fernank2 
https://voltaconsolar.com/blog/knowledge-base/how-to-resolve-error-61-with-pylontech-lithium-batteries-and-voltacon-solar-inverters/
"Error 61 means that the batteries are not communicating with the inverter, and the BMS is not installed successfully. There are a few reasons why"

Fernank2, 2023-09-15:
> @Fernank2 https://voltaconsolar.com/blog/knowledge-base/how-to-resolve-error-61-with-pylontech-lithium-batteries-and-voltacon-solar-inverters/ "Error 61 means that the batteries are not communicating with the inverter, and the BMS is not installed successfully. There are a few reasons why"
> 
> This is an example cable for an RS232 if I understand correctly. You have to build an rs485 cable with the wiring I specified. If the inverter can RS485
> 
> [image]

I insist, my error is error 60, I don't have error 61, okay? 60 YES—61 NO.

Fernank2, 2023-09-15:
on the other hand moscito1010. Can you send a screenshot of your configuration of all the diybm sections? It also sends a capture of the current parameters of your inverter. parameter 3, 26, 27 and 29. please

moscito1010, 2023-09-15:
https://powerforum.co.za/topic/8914-axpert-mks-5k4kw-error-60/

Fernank2, 2023-09-17:
> https://powerforum.co.za/topic/8914-axpert-mks-5k4kw-error-60/ Fernank2 ***@***.***> schrieb am Fr., 15. Sept. 2023, 15:47:
> […](#)

I have already done all that. that is not my case because I am emulating phylontech and my error only happens with phylontech485.
Does it really work for you?
@moscito1010 , you say it works for you… can you send a screenshot of your configuration of all the diybm sections? It also sends a capture of the current parameters of your inverter. parameter 3, 26, 27 and 29. please

moscito1010, 2023-09-18:
That won't help you.
I think I have a better solution for you.

You need a “Men in the Middle” in your case “Solar Assistant.

Using a USB-CANBUS adapter, you connect “Solar Assistant” to the DIYBMS via CANBUS. Set USE on the Voltronic Inverter and connect the inverter to the “Solar Assistant” with a USB-RS232 cable.

“Solar Assistant” takes over the charging control of the inverter. You only set the normal parameters on the inverter. Shutdown Voltage, Back to Grid Voltage, Max Battery Voltage and the Solar/Battery/Utility (SBU) mods
The “emulated BMS” takes care of the rest

Please contact Solar Assistant support.
I'll use the same solution myself, I'm just waiting for my USB-Canbus adapter. I don't want to use patched firmware, the risk of the branch not being maintained as intensively is too great for me.

It's important to me that I get support from the original author, I can't expect him to also care about whether a fork is OK. Stuart knows his "baby" and I tend to trust the original.

Fernank2, 2023-09-19:
The images of a system that you say works for you will help me, compare them with mine, which does not work. I think it doesn't work for you either. thanks for your answers. Let's see if Ruza responds. I'll ask Stuart too.

moscito1010, 2023-09-19:
Here are a few pictures of my settings.
I would like to point out that I also use an active equalizer that starts working at 3.45v and stops working at 3.4v.
The settings are from my test system which is connected to my Axpert Max 7.2 with RS485 Pylontech.

Modules:
[image]

Rules:
[image]

Settings:
[image]

Charge/Discharge
[image]

Voltage and Amperemeter:
[image]

moscito1010, 2023-09-19:
Here is the possible solution:
You can use a “men in the middle”.
The SA (Solar Assistant) software helps you use devices that cannot communicate directly.
The BMS and the Voltronic inverter are connected to the SA Rasperry pi using suitable adapters. In this case a USB-CANBUS adapter and a USB-RS232 (FTDI) adapter.
On the inverter you then set the battery type to USE because SA takes over the control with its emulated BMS.
The DIYBMS tells SA what it wants and SA controls the inverter.

I just want to switch to this solution myself and have therefore asked SA support.
  SA offers a USB-CANBUS cable with the appropriate programming for various inveter on its website. The cable is not expensive.
The software is not expensive and it is a lifetime license at no additional cost. The disadvantage is that SA uses a Raspberry completely, you can't install anything else. But since I have a few more Raspberries, it's not that dramatic and the Raspi's are available and affordable again.

But I think we should explore this in a different way. That no longer has anything to do with diybsm support.
pvbeere@tutanota.de or even better Telegram: https://t.me/DIYBMSv4