stuartpittaway/diyBMSv4ESP32 #229: What can the SSR tolerate?

moscito1010, 2023-07-28:
What can the SSR tolerate?
If I understood correctly, I switch an external voltage with the SSR port (REL3 or 4). I want to control a thick 500A SSR with it. The external SSR needs 3-32V/18mA, how much voltage can I apply to REL3 or REL4?

virtuvas, 2023-07-29:
I'm using them to turn on and off a 500A blues SSR. No problems.

stuartpittaway, 2023-07-29:
Hi @moscito1010, the answer unfortunately depends....

There have been about 3 different chips used for those relay ports (RELAY3 and 4), so you need to check the datasheet for accurate information.

The design on the "master" branch uses the Panasonic AQY282SX [datasheet](https://datasheet.lcsc.com/lcsc/1912111437_PANASONIC-AQY282SX_C250827.pdf)

This relay can handle 60V and 500mA (max) 

However, please check the chip you have!