stuartpittaway/diyBMSv4ESP32 #231: daily Ah out/in view wrong

buddhafragt, 2023-08-08:
Hello,
the daily Ah out/in view is wrong
[image]

stuartpittaway, 2023-08-09:
Hello @buddhafragt 

Can you tell me some more about the system you are running - are you using the latest versions of code on both controller and the external current monitor/shunt (unless using the onboard one) ?

Does this happen all the time, or is this the first power on?

If you use the "reset daily amp hour" button, does the value correct to zero?

[image]

Have you been using the set S.o.C function?

stuartpittaway, 2023-08-10:
Ok, if the reset option doesn't work, you need to update the shunt firmware/code. You are running with a mix of incompatible versions

buddhafragt, 2023-08-10:
Thank you Stuart, I will do it!

stuartpittaway, 2023-08-15:
Did you get chance to test this?

closed 2023-09-12 by stuartpittaway