stuartpittaway/diyBMSv4ESP32 #232: INA229 - SOC is always lower than the value it is set to,

delboy711, 2023-08-18:
Running current Master 5b7135f

When Utilities>Set State of charge is used to set SOC Whatever SOC level is chosen the SOC will always be less than the figure selected so long as Charge efficiency factor  is less than 100%.

The reason is because Charge efficiency factor is not taken into account when setting SOC, but **is**  taken into account when comparing milliamphour_in  to milliamphour_out to calculate current SOC.

At first sight this is no big deal, barely an inconvenience . 

Until one considers what happens when the battery is becoming full and the test conditions of CalculateAmpHourCounts().
Namely when battery voltage has reached registers.fully_charged_voltage  and tail current has dropped below its limit for three minutes.

At that point SetSOC(10000) is called to force SOC to 100% and  the inverter should stop charging. **But**, this bug means the SOC is actually set to something below 100% and so it will continue charging for ever. SOC will creep up and then be reduced back down again ad infinitum.

The parameters used in my system are:-
Battery Capacity 105Ah
Charge efficiency 99%

stuartpittaway, 2023-08-18:
Hi @delboy711 thanks for the issue.  This is something that needs careful thought on the best resolution due to the way SoC is calculated by the shunt.

I've also implemented "float" controls in the next release, so the charger would be shut off after 99% after its completed its Absorb and Float time periods.

delboy711, 2023-08-18:
The mixture of floating point and integer arithmetic warped my brain a bit, but eventually I came up with this.

It works for me for all values of SOC
```
void CurrentMonitorINA229::SetSOC(uint16_t value)
{
    // Assume battery is fully charged
    milliamphour_in = lround(100000.0*(float)registers.batterycapacity_amphour/(registers.charge_efficiency_factor));
    // And we have consumed this much...
    milliamphour_out = (uint32_t)((1.0 - ((float)value / 10000.0)) * (1000.0*(float)registers.batterycapacity_amphour));

    // Zero out readings using the offsets
    milliamphour_out_offset = milliamphour_out;
    milliamphour_in_offset = milliamphour_in;
}
```

stuartpittaway, 2023-08-22:
Thanks for this, might be a few days before I can test and see what impact this has on the system as there are a few rules and other calculations based on SoC.

stuartpittaway, 2024-03-07:
This change has been implemented in commit 3fc920e27a0c75ce837ca114cd9008de2d163eb4