stuartpittaway/diyBMSv4ESP32 #234: error: pins_arduino.h: No such file or directory

francklefevre, 2023-08-18:
Dear All,

I face a strange problem while trying to compile the Controler's source from my Windows box :
Note that I'm not familiar with platformIO. May be I've forgotten something obvious but the build of the CellModule was perfect.

**********************************************************************
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.21.3+sha.9794ba2
|-- SerialEncoder @ 0.1.0+sha.f909d31
|-- TFT_eSPI @ 2.5.31+sha.caa1076
|-- crc16
|-- FS @ 2.0.0
|-- LittleFS @ 2.0.0
|-- SD @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
Building in release mode
Compiling .pio\build\esp32-devkitc\src\CurrentMonitorINA229.cpp.o
Compiling .pio\build\esp32-devkitc\src\HAL_ESP32.cpp.o
Compiling .pio\build\esp32-devkitc\src\PacketReceiveProcessor.cpp.o
Compiling .pio\build\esp32-devkitc\src\PacketRequestGenerator.cpp.o
In file included from C:/Users/franc/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:79,
                 from C:/Users/franc/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from include/CurrentMonitorINA229.h:37,
                 from src/CurrentMonitorINA229.cpp:30:
C:/Users/franc/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory
**********************************************************************

Thank you for your help.

Franck

stuartpittaway, 2023-08-19:
Unless you are planning to make code changes, I recommend using the precompiled ones in the release section

francklefevre, 2023-08-20:
Dear Stuart,
Of course I plan to make code changes, and may be 🤞  to become an active contributor to the project !
Bst rgds.

Franck

stuartpittaway, 2023-08-20:
Fantastic!

Platform io should have pulled all the dependent files, did you clone the git repo using GitHub desktop?

francklefevre, 2023-08-20:
Yes, I cloned the project by : 
      git clone https://github.com/stuartpittaway/diyBMSv4ESP32.git
...and I tested it 3 times, deleting the all directory between each test.
Do you know in what package/directory the file [pins_arduino.h] is supposed to be ?
I can check if the concerned directory is there.

stuartpittaway, 2023-08-20:
It's not part of my project, it's the core Arduino framework.

francklefevre, 2023-08-21:
I've found a fix to this issue.
For an unknown reason, the directory:  
   .platformio\packages\framework-arduinoespressif32\variants
and its content has not been installed by platformio. Even when I tried to uninstall/reinstall this platform's package.
By copying on the windows system the version that was fully installed on my linux box, everything compiles and runs fine !