stuartpittaway/diyBMSv4ESP32 #237: Frequent controller restarts.

moscito1010, 2023-09-04:
Hello Stuart.
Since yesterday I have had one of my 3 inverters back from repair. I use the RS485 branch.
The controller restarts quite often without me being able to understand why it is doing this.
I had the serial output written to two text files for you so you can take a look at the log.
It seems as if this only happens when the controller has to work and the dynamic charging is very active and the charge controller is often readjusted. But this is not a forensic observation but rather a guess. When the memory is full and the charging current
is low, the spitting stops again.
[diybms2.zip](https://github.com/ned-kelly/docker-voltronic-homeassistant/files/12514437/diybms2.zip)
[diybms.zip](https://github.com/ned-kelly/docker-voltronic-homeassistant/files/12514438/diybms.zip)
My Controller is the unit from your Patreon sale out and the Modules are the V4.50

Here Pic's from my Hardware
[image]
[image]
[image]

moscito1010, 2023-09-04:
And because it's so beautiful... :-)
I tried to reproduce the error and noticed that the controller keeps restarting when I switch on the export to an influxdb.
Logs will follow tomorrow.

stuartpittaway, 2023-09-05:
Hi @moscito1010 - I've not used the RS485 version, so can't directly support any problems you have with it, it was created by [ruza87](https://github.com/ruza87)

moscito1010, 2023-09-05:
oh damn, right.

stuartpittaway, 2023-09-05:
Did you compile the code yourself?  

If you did, look in main.cpp for a line like...

```xTaskCreate(sdcardlog_outputs_task, "sdout", 3200, nullptr, 0, &sdcardlog_outputs_task_handle);```

and change the "3200" to a higher number (3400) and upload the firmware again.

moscito1010, 2023-09-08:
Since the topic of RS485/Pylontech really "triggers" me, I'll just ask. Would it be possible for you to contact me by email? pvbeere@tutanota.de
best regards

ruza87, 2023-09-22:
Hi @moscito1010, thank you for your feedback. I agree with @stuartpittaway, from the log it seems like some memory issue when logging to SD card. I'll analyze the logs in detail next week. Meanwhile, can you please remove the SD card (to stop the log being written) and see if it helps?

ruza87, 2023-09-25:
I've just merged the newest diyBMS sources into RS485 branch, can you please report if that helped?

Was there any improvement after increasing the stack size for SD task as suggested by @stuartpittaway?