stuartpittaway/diyBMSv4ESP32 #239: Lost WIFI connection that is not restored until the controller is restarted

chapulino, 2023-09-11:
**Describe the bug**
Quite often lately, up to three times a day, the Wi-Fi connection between the controller and the Wi-Fi network is lost, and the controller cannot be accessed. This does not influence its operation since it is communicated by Modbus with a CERBO, but the controller stops sending monitoring data via MQTT.
It surprises me that the controller itself, once the Wi-Fi connection is lost, does not try to reestablish it again.

**Hardware/Software Versions**
Controller version (from PCB): 4.2
(below can be obtained from the "About" page in the controller web interface)
Processor: ESP32
Version: [f6e94fbd18cf7803e818459735021730541ce20b](https://github.com/stuartpittaway/diyBMSv4ESP32/commit/f6e94fbd18cf7803e818459735021730541ce20b)
Compiled: 2023-09-08T14:51:55.713Z

**To Reproduce**
It is an error that cannot be caused, it appears occasionally without my having been able to establish a cause for it.
It could be something within the function to automatically re-establish a Wi-Fi communication in the event that it has been lost for any reason.

alxbauer, 2023-09-12:
Same problem here after restarting my router last night.

With the preview firmware version from June 14, the problem never occurred. I think the problem only occurs with the latest firmware version.

stuartpittaway, 2023-09-12:
The controller code does attempt to reconnect to the WIFI, but something has obviously changed/not working.  The IDF firmware in the latest release has been upgraded to 4.4.5 - which could be the underlying issue.

To resolve this I need:

1. The signal level from the TFT display (dBM) value
2. The USB serial output from the ESP32 when this problem occurs

chapulino, 2023-09-12:
I'm sorry I'm not very useful, first of all I wouldn't know how to measure the DBM value. And I don't have the controller in a place where I can easily get the serial USB Ouput

stuartpittaway, 2023-09-12:
The DBM value is shown on the TFT screen (bottom right before the time) - mine reports -74dBM - its the WIFI signal strength.

stuartpittaway, 2023-09-12:
I've just pushed another release up which changes to ESP Arduino framework 2.0.12 - version 2.0.11 had a bug in it which caused lots of memory to be used.  This could be a reason why WIFI became unstable.

New release is here. Please try this one and see if the WIFI problem is resolved.

https://github.com/stuartpittaway/diyBMSv4ESP32/releases/tag/Tag-2023-09-12-10-49

chapulino, 2023-09-12:
Updated, I will comment in a couple of days if the problem has stopped occurring. Regarding the strength of the WIFI signal, it has a value of -88dbm

chapulino, 2023-09-13:
20 hours and the Wi-Fi communication has not been lost, or if it has been done punctually, it has been reestablished without being able to be seen. Stuart, what you did worked. Thank you.

JochenSchmidt, 2023-09-14:
@chapulino 
What happens if you turn off the WiFi and turn it back on again after a few minutes? Is it reconnecting?

JochenSchmidt, 2023-09-14:
@chapulino 
You wrote: This does not influence its operation since it is communicated by Modbus with a CERBO, but the controller stops sending monitoring data via MQTT

Is that right or do you mean diyBMS CAN-bus communication with Cerbo?

chapulino, 2023-09-14:
> What happens if you turn off the WiFi and turn it back on again after a few minutes? Is it reconnecting?

I have not performed that test, although I imagine that with Stuar's correction, the system reconnects automatically.

JochenSchmidt, 2023-09-15:
> There is a Modbus cable communication with the Victron system.

Sorry chapulino, I don´t understand that..... you mean that you have a modbus cable connection from diyBMS to Victron cerbo?
But the diyBMS doesn´t have Modbus.....only CAN-Bus to victron CAN-Bus-Battery.
I´m using the diyBMS CAN-Bus to Victron Battery CAN-Bus and there is also no communication break if WiFi is down.
Modbus I use with Cerbo GX to read and write values by ioBroker. diyBMS-MQTT I also use  whis ioBroker-Adapter to read values and visualize in ioBroker VIS. diyBMS-MQTT is over WiFi.

chapulino, 2023-09-15:
Sorry for the confusion, I was actually referring to cambus

HerrFrodo1, 2023-09-25:
@chapulino Hi chapulino, have you been able to try out what happens when you turn your wifi off and on again? Or do you only switch it on again after 5 minutes? That would be a great help for me - thanks :-)

chapulino, 2023-09-25:
If I turn the controller off and on again, and there is a sufficient Wi-Fi signal, it connects and does not present any problem. The problem described was solved with the update. Now I'm waiting for an ESP32 with an antenna to improve the WiFi signal

HerrFrodo1, 2023-09-25:
@chapulino Thank you very much!
Did you wait a few minutes?
After a few minutes my ESP32 won‘t connect  to wifi again :-(
I also use IOBroker as MQTT-Server. 
Maybe my Problem is the MQTT connection.

HerrFrodo1, 2023-09-25:
Wifi has -40dBm. 
Yes, same firmware. 
I made a little test and deactivated MQTT in diyBMS and ioBroker. 
Thats it! MQTT is the murder!
Without MQTT, wifi is 30sec after presence up on diyBMS.

chapulino, 2023-09-25:
At the time it was reported that having the two integrations activated at the same time, MQTT and Influx, could cause problems, I suppose that is not the case, but to investigate more causes

stuartpittaway, 2023-09-26:
Another user on Patreon has reported issues with wifi not restoring correctly - this also seems to be tied into having MQTT enabled.

Needs further investigation, as I think the wifi and MQTT libraries are not being shut down correctly when WIFI is lost.

GregoryFi, 2023-11-06:
Hello - I just performed a test with Release-2023-10-30-10-43.
After about 15hours I get the following:
E (6666795) MQTT_CLIENT: Error transport connect
E (6666799) diybms-mqtt: MQTT_EVENT_ERROR
E (6666803) diybms-mqtt: Last err no string (Success)
I (6666808) diybms-mqtt: MQTT_EVENT_DISCONNECTED
E (6666806) diybms-mqtt: MQTT enabled. WIFI not connected
I (6667900) diybms-mqtt: Connect MQTT
E (6667900) TRANSPORT_WS: esp_transport_ws_init(621): Memory exhausted
E (6667901) MQTT_CLIENT: create_client_data(765): Memory exhausted
E (6667904) diybms-mqtt: mqtt_client returned NULL
E (6668264) diybms-mqtt: MQTT enabled. WIFI not connected
E (6670313) diybms: Connect to the Wifi AP failed
E (6670330) esp-tls: [sock=51] connect() error: Host is unreachable
E (6670331) TRANSPORT_BASE: Failed to open a new connection: 32772

stuartpittaway, 2023-11-06:
Hi @GregoryFi, how are you testing this - disabling WIFI on the access point?

It looks like the controller is staying "up" rather than rebooting/running into issues.  Did the WIFI reconnect successfully after the test?

I can see the MQTT has run into problems with memory allocation.

GregoryFi, 2023-11-06:
Hi Stuart - I connected a laptop with USB through serial port with the controller and Putty to log, and left it running until I saw I could no longer access the controller through local ip. Wifi was never disabled, but also never restored until I rebooted the controller (cutting the power).

stuartpittaway, 2023-11-06:
Ok @GregoryFi - I can see errors similar to ```Connect to the Wifi AP failed``` can you confirm RSSI (wifi signal strength) to the ESP32 - this is in the lower right of the TFT display if you have that fitted.

GregoryFi, 2023-11-06:
Eh, not really ... I removed the TFT as it caused issues in the past. But I can recall it was around -50 to -54dB when I had it mounted. I did have similar issues with the previous controller type (using ESP8266) - it was the main reason I wanted to upgrade my controller; and sometimes the ESP32 in my OPenEVSE disconnects, but it always reconnects. The main difference in wifi configuration is I have attributed a fixed IP to the controller (as I thought DNS was causing the issues), and the ESP32 in the OpenEVSE switches ip dynamically almost every time it resets ...

stuartpittaway, 2023-11-06:
Ok, I don't really see the same issues you are experiencing - it looks like a poor quality WIFI connection, hence all the errors.  The comments above in this issue relate to people with low signal strength (-88dBm) or who are deliberately switching off WIFI access points on a schedule/overnight.

Could this be an underlying issue with your wifi router/access point?

GregoryFi, 2023-11-06:
Probably, but I have difficulties understanding the difference between poor quality and low signal strength : the main issue is it doesn't reconnect automatically, The issues I had with the TFT installed was it would reboot all the time, ... in this case it seems an automatic reboot when the wifi connection is lost would be ideal: would there be any inconvenience in coding this ?

stuartpittaway, 2023-11-06:
> The issues I had with the TFT installed was it would reboot all the time, ... in this case it seems an automatic reboot when the wifi connection is lost would be ideal: would there be any inconvenience in coding this ?

The TFT issue should have been resolved if you want to try that again - there appears to be some difference between various revisions of the ESP32 hardware/chips which caused a problem.

It wouldn't be a good idea if the controller rebooted on WIFI being lost - if WIFI goes down for a long period of time, the controller would be in an endless reboot loop - and all the time, your battery is at risk of damage/over charge.

The quality of the wifi signal, rather than just signal strength if very important - the two are often linked, but it seems like a number of data packets get lost when communicating with MQTT, or for DNS lookups.  Do you have any firewalls or other blocking devices on the network?

GregoryFi, 2023-11-06:
Blocking devices as such not but I am using a PLC to create wifi in the garage as there is no repeater that is strong enough.
Good point about the loss of wifi, but in that case can you confirm that CANBus communication is independent from wifi ? @chapulino says his communication with Victron is upheld, but @JochenSchmidt says without wifi, CAN is also down.

chapulino, 2023-11-06:
I confirm, at least in my case, that a Wi-Fi outage causes me to lose mqtt data communication, but cambus communication is not affected. Although I occasionally lose statistics, the important thing about DIYBMS Victron control and communication is not affected. In my case, I significantly improved the signal (with a WIFI repeater) and the problem already happens very occasionally.

stuartpittaway, 2023-11-06:
> CAN is also down.

There isn't any dependency on WIFI and the CANBUS communication - they both operate independently.

Bingo2023, 2023-11-18:
seems these unstable Wifi connection come from Platform SDK package, in most cases with mash Networks.
The ESP32 will connect from time to time to very far distance AP and stucks with the weak connection.

For you reference see below and check for "BSSID" it seems to help in some cases.

https://github.com/emsesp/EMS-ESP32/issues/1321

chapulino, 2023-11-19:
The truth is that this could explain what happens to me, because in my case I operate with a mesh Wi-Fi network.

stuartpittaway, 2023-11-19:
Do you folks use Fritzbox ?

Bingo2023, 2023-11-19:
> Do you folks use Fritzbox ?

yes

Bingo2023, 2023-11-19:
I spend some hours to check situation with different Versions.

Release
10.30.	-> Connection to Main AP OK; -> no connection to slave AP / no main screen -> keeps resetting
14.10.	-> Connection to Main AP OK; -> no connection to slave AP / no main screen -> keeps resetting
12.09.	-> no Connection to Main AP ! -> keeps resetting
08.09.	-> Connection to Main AP OK; -> connection to slave AP OK -> reset ever 20min
04.17.	-> Connection to Main AP OK; -> connection to slave AP OK -> runs unstable but self recovering
I used the Feb 2023 Versions and they had none of these problems.

Summary:
the Wifi Problem becomes worst with each version.

Bingo2023, 2023-11-19:
[Release-2023-10-30-10-43](https://github.com/stuartpittaway/diyBMSv4ESP32/releases/tag/Tag-2023-10-30-10-43)
Attached is the Logfile from Serial Output
[Release-2023-10-30-10-43_log.txt](https://github.com/stuartpittaway/diyBMSv4ESP32/files/13403290/Release-2023-10-30-10-43_log.txt)

stuartpittaway, 2023-11-19:
> Summary:
> the Wifi Problem becomes worst with each version.

That's a little harsh!!  What normally happens is that the underlying firmware (esp32 libs) get updated to a newer version, which in theory should fix previous bugs, but I agree doesn't always.

Have you tried this version?

https://github.com/stuartpittaway/diyBMSv4ESP32/tree/wifi_mqtt_fixes

There seems to be a common issue with Fritzbox and the esp32, I have some other users running into similar problems.

I don't use that, and have uptime of more than 20 days straight. I also use the esp32 D variant with external aerial.

Bingo2023, 2023-11-19:
didn't intend to say something bad on this wonderful project.
Obviously the root-cause is in the libs to find.

General I agree, it is best to have updated Libs, but this bug in a essential function as wifi cause me already many days and hours of frustration (also on other projects). If I read in other forums there are tons of people struggling with it....
Without Wifi, the ESP32 is a useless peace of electronic sh**.

Bingo2023, 2023-11-19:
> Have you tried this version?
> 
> https://github.com/stuartpittaway/diyBMSv4ESP32/tree/wifi_mqtt_fixes

are there compiled versions or do I need to compile myself? Will tray that brach..

stuartpittaway, 2023-11-19:
Yes there are precompilee versions, take a look at the GitHub actions section, and then the artifacts. You will need to be logged into GitHub to view those.

Bingo2023, 2023-11-19:
> https://github.com/stuartpittaway/diyBMSv4ESP32/tree/wifi_mqtt_fixes

this branch works much better.
Wifi connect and controller boots up. I saw a reset already but seems that BMS is coming back and establish a fresh & working wifi connection. I will check this version in next days.