stuartpittaway/diyBMSv4ESP32 #240: Powervalue fron INA229 when current is near Zero

midopple, 2023-09-12:
**Describe the bug**
If the measured current from the INA229 (internal current sensor) is close to 0 then a power of > 70 W is still displayed, since the power is calculated in the INA229 chip the question is whether the error is in the hardware or in the display.

If the current value is >5A, the measured power is very good.
[image]

**Hardware/Software Versions**
Controller version (from PCB): 4.6
(below can be obtained from the "About" page in the controller web interface)
Processor: ESP32
Version: latest Masterbrach from Github
Compiled: 10.9.2023

**To Reproduce**
Steps to reproduce the behavior:
Use internal Currentsensor and check the Powervalue when Current < 1A

stuartpittaway, 2023-09-12:
Thanks for reporting this, very strange!

I'll take a look as the value seems to be x10 out (should be 7.08W)

midopple, 2023-09-12:
Hi,

I have taken some values from the log files and calculated the power and compared it with the value from the IC

on the X-axis is the current from +50 to +33A on the Y-axis the power and on the secondary axis the error in watts

It looks like the error occurs only in the range of +/- 4 A around the zero point.

[image]

Table

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/midopple/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/midopple/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

voltage | current | Power from   INA229 | Power calculate | Error [W]
-- | -- | -- | -- | --
52,136 | -50,9853 | 2662,5964 | -2658,169601 | 4,4267992
52,026 | -50,7162 | 2642,8625 | -2638,561021 | 4,3014788
52,827 | -19,1712 | 1013,3875 | -1012,756982 | 0,6305176
52,572 | -10,0389 | 527,979 | -527,7650508 | 0,2139492
52,987 | -7,2633 | 384,9525 | -384,8604771 | 0,0920229
52,95 | -7,1386 | 378,077 | -377,98887 | 0,08813
52,672 | -6,6131 | 348,4155 | -348,3252032 | 0,0902968
53,014 | -5,9867 | 317,439 | -317,3789138 | 0,0600862
53,042 | -4,4362 | 235,3415 | -235,3049204 | 0,0365796
52,714 | -3,5278 | 185,9935 | -185,9644492 | 0,0290508
52,739 | -2,2511 | 121,958 | -118,7207629 | 3,2372371
52,747 | -0,9708 | 84,2115 | -51,2067876 | 33,0047124
55,856 | -0,17 | 81,7905 | -9,49552 | 72,29498
55,921 | -0,06 | 81,1345 | -3,35526 | 77,77924
56,723 | 0,4622 | 75,557 | 26,2173706 | 49,3396294
56,712 | 0,7966 | 80,0775 | 45,1767792 | 34,9007208
52,86 | 1,6891 | 119,9235 | 89,285826 | 30,637674
52,878 | 2,4553 | 129,8145 | 129,8313534 | -0,0168534
52,925 | 3,7361 | 197,712 | 197,7330925 | -0,0210925
53,074 | 6,6278 | 351,7245 | 351,7638572 | -0,0393572
53,13 | 7,7017 | 409,1395 | 409,191321 | -0,051821
53,27 | 8,6816 | 462,389 | 462,468832 | -0,079832
53,434 | 10,8691 | 580,654 | 580,7794894 | -0,1254894
53,581 | 13,3095 | 712,9255 | 713,1363195 | -0,2108195
53,795 | 15,2305 | 819,0445 | 819,3247475 | -0,2802475
53,906 | 19,0834 | 1028,2455 | 1028,70976 | -0,4642604
53,971 | 20,51 | 1106,3765 | 1106,94521 | -0,56871
54,022 | 22,6191 | 1221,218 | 1221,92902 | -0,7110202
54,061 | 25,2087 | 1361,9845 | 1362,807531 | -0,8230307
54,08 | 26,2312 | 1417,5549 | 1418,583296 | -1,028396
54,135 | 31,0605 | 1679,98 | 1681,460168 | -1,4801675
54,152 | 32,9797 | 1784,3474 | 1785,916714 | -1,5693144

</body>

</html>

Linusten, 2023-09-17:
can confirm that issue on my side too, controller is gen 4.5 with ina229 and latest firmware (Victron 500A shunt).
[image]

Chris87-ux, 2023-09-18:
Hi together, seems to be the same issue as https://github.com/stuartpittaway/diyBMSv4ESP32/issues/103.
I also have permanently the same issue.
/Chris

moscito1010, 2023-09-29:
I connected a Raspi with Solar Assistant to the inverter and can compare the values quite well. The measured value that is output serially on the Voltronic is almost exactly the same as with the BMS at high loads.
I never noticed that there was a difference at low loads.
Solar Assistant is currently showing me that 100w is going into the battery because I have set it so that 2A DC goes into the battery when the inverter switches to grid.
mqtt is from the inverter
[image]
[image]

stuartpittaway, 2023-09-30:
It's still on my to-do list!

stuartpittaway, 2023-10-02:
Researching this problem, the INA229 chip provides in-built power calculation (datasheet is [here](https://www.ti.com/document-viewer/INA229/datasheet/GUID-25B37629-8261-48D8-BE56-62622C36C345#TITLE-SBOSA20INA229))

This is the code which reads the 24 bit [power register](https://www.ti.com/document-viewer/INA229/datasheet/GUID-25B37629-8261-48D8-BE56-62622C36C345#TITLE-SBOSA20INA229) and returns it to the controller code.  The register is described as:
* Calculated power output.
* Output value in watts.
* Unsigned representation. Positive value.
* Value description under [Section 8.1.2](https://www.ti.com/document-viewer/INA229/datasheet/GUID-C649C55C-F10D-4678-9D22-5B327520E0DA.html#GUID-C649C55C-F10D-4678-9D22-5B327520E0DA).

```
    // Calculated power output.  Output value in watts. Unsigned representation. Positive value.
    float Power()
    {
        // POWER Power [W] = 3.2 x CURRENT_LSB x POWER
        return (float)spi_readUint24(INA_REGISTER::POWER) * (float)3.2 * registers.CURRENT_LSB;
    }
```
The code calls ```spi_readUint24``` which is the only place where this 24 bit register read is used.

```

// Reads 3 bytes from SPI bus, and returns them in a uint32 (lower 24 bits)
uint32_t CurrentMonitorINA229::spi_readUint24(INA_REGISTER r)
{
    SPI_Ptr->beginTransaction(_spisettings);
    digitalWrite(chipselectpin, LOW);
    SPI_Ptr->write(readRegisterValue(r));
    uint8_t a = SPI_Ptr->transfer(0);
    uint8_t b = SPI_Ptr->transfer(0);
    uint8_t c = SPI_Ptr->transfer(0);
    digitalWrite(chipselectpin, HIGH);
    SPI_Ptr->endTransaction();

    uint32_t value = ((uint32_t)a << 16) | ((uint32_t)b << 8) | ((uint32_t)c);

    return value;
}
```

The above code, reads the 3 register bytes (24 bit in total), and assembles into a uint32 value before being returned.

I cannot seem to find a code error in the above - I'm wondering if the error is creeping in during the multiplication of several of the "float" values?  Anyone help with this?

midopple, 2023-10-05:
Hi,

I have done some tests on my System.
I have disconnected all consumers from the battery and connected only one resistor, this shows that the power measurement is correct, see screen shot.

[image]

When I connect the Multiplus II to the battery, I have the error of the power when the current is below 1.2A.

From my point of view, the error is not in the software.
Perhaps the Multiplus generates interference or current peaks at low power which the INA229 chip interprets incorrectly.

stuartpittaway, 2023-10-05:
> From my point of view, the error is not in the software.
> Perhaps the Multiplus generates interference or current peaks at low power which the INA229 chip interprets incorrectly.

Are the voltage and current readings correct?  The INA chip is configured to perform multiple "oversamples" of the voltage and current as part of its measurement - we could try reducing/increasing the number of samples to see if that helps.

midopple, 2023-10-05:
This is a good idea, I will adjust the settings in the firmware and check if this makes a difference.

The voltage measurement which i check with a multimeter was correct.

For the current measurement, my clamp ampere meter is too inaccurate to have a comparison here.

midopple, 2023-10-16:
I have tested some settings for the ADC (samplerate and AVG), see table.

If you increase the AVG value towards 256 or 512
the measuring error is a little bit smaller, but it is still about 70W at 0A. 

Unfortunately I had with the long sample time (> 1sec) again and again error that the ESP restarts and it has only brought a minimal improvement.

Currently I do not quite understand what the Victron Multiplus 2 at the small currents makes that this measurement error occurs, especially when current and voltage are measured correctly and only the power is wrong.

Maybe I can check the voltage at the shunt with an oscilloscope and isolation amplifier.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/midopple/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/midopple/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Bus VOLT | Current |   |   |   |  
-- | -- | -- | -- | -- | --
VBUSCT | VSHCT | AVG | R_ADC_CONFIG | Sampletime | Note
0x7h (4120 µs) | 0x7h (4120 µs) | 0x4h (128) | 0xFFD4 | 530 ms | Original Settings,   Error when Current < 1A ~ 80 W
0x7h (4120 µs) | 0x7h (4120 µs) | 0x5h (256) | 0xFFD5 | 1054 ms | Error when Current   < 1A ~ 70 W, samplerate low > 1sec
0x7h (4120 µs) | 0x7h (4120 µs) | 0x5h (512) | 0xFFD6 | 2109 ms | Error when Current   < 1A ~ 70 W, error at Startup
0x7h (4120 µs) | 0x7h (4120 µs) | 0x3 (64) | 0xFFD3 | 263 ms | Error get bigger ~ 95   W
0x5h (1052 µs) | 0x5h (1052 µs) | 0x3 (64) | 0xFB53 | 16ms | Error ~ 120 W
0x5h (1052 µs) | 0x5h (1052 µs) | 0x4h (128) | 0xFB54 | 134ms | Error ~ 115W
0x5h (1052 µs) | 0x5h (1052 µs) | 0x5h (256) | 0xFB55 | 270ms | Error ~ 115W
0x5h (1052 µs) | 0x5h (1052 µs) | 0x6h (512) | 0xFB56 | 530ms | Same as Original,   Error at current < 1A ~ 80 W
0x4h (540 µs) | 0x4h (540 µs) | 0x6h (256) | 0xF916 | 138ms | Error ~ 115W

</body>

</html>

At the moment I use this workaround, if the powervalue is smaler then 200W
calculate from Volate and current the Power and replace the Value

```
void CurrentMonitorINA229::TakeReadings()
{
    float help_pow_val = 0;
    
    voltage = BusVoltage();
    current = Current();
    power = Power();

    help_pow_val = voltage * current;
    if(help_pow_val < 0) help_pow_val = help_pow_val * (-1.0);

    if(help_pow_val < 200) { power = help_pow_val; }

    temperature = DieTemperature();
    SOC = CalculateSOC();

    CalculateAmpHourCounts();

    ESP_LOGD(TAG, "V=%.4f, I=%.4f, P=%.2f", voltage, current, power);
}
```

stuartpittaway, 2023-10-17:
I'm starting to wonder if this is a fault on the INA chip - alternatively it could be a rounding issue in the code.

Is it worth ignoring the power reading completely from the chip and simply do voltage * current ?

Chris87-ux, 2023-10-17:
A while ago (arround 12 month) I added the same but and did the same research, unfortunately I was not able to find this bug anymore.

stuartpittaway, 2023-10-17:
@midopple have we factored in the power loss from the multiplus?  Is it particularly "bad" at low currents and its efficiency drops?

My SOFAR inverter appears to have about 10% loss between its readings and the battery current readings - but it is fairly constant.

midopple, 2023-10-17:
Hi,

The Multiplus has about 11W losses, see table

[image]

From the structure:
The shunt is located between the battery and Multiplus and measures everything that is consumed from the battery, including the power loss.

The values voltage and current are transmitted via CAN BUS to the GX device.

The display on the Victron system is correct because Victron calculates the power from the current and the voltage value from CAN. The Intern current Sensor (Multiplus) show the same Value as the current from the Shunt 

The Victron system regulates correctly, only the display on the ESP32 website is not correct, but this is only a visual problem.

The Multiplus tries to reduce the losses at low power (AES).
I have deactivated but maybe something else is active here.

I think there is no rounding errors in the software of the ESP, I have tried here different variants of the calculation.
The same result was always output.

midopple, 2023-11-02:
Hi

Thanks a lot, the new version of the firmware works great for me.

Should I close the issue ?

closed 2023-11-02 by stuartpittaway