stuartpittaway/diyBMSv4ESP32 #243: All In One temperature issues.

delboy711, 2023-09-28:
Not sure if you are ready to take issues with the AllInOne board yet so feel free to close this issue.

I have built an AllInOne and am experiencing two issues with how the temperature sensors operate.

1/  If no balance board is fitted the ESP32 controller will show a continuous under temperature alarm.
This is because the on board sensor is attached to Cell 0 External Temperature, and Internal Temp is not connected.

2/ If the AllInOne board is located in a position where it could be colder than the battery cells (my cells are well insulated), and the cells are able to be warmed with a heating pad, then the relay to warm the cells would activate too soon.

Removing the sensor TH6 would probably be a work around for the second issue, but would it not be better for the onboard sensor to be assigned as the 'Internal' sensor of Cell 0? That would also resolve the Under Temp alarm.

Otherwise the AllInOne is working brilliantly.   Good Job!

stuartpittaway, 2023-09-29:
Hi @delboy711 glad to hear the board is generally working well for you.

On the temperature stuff...

The three external temperature sensor sockets J11/J12/J13 are mapped to EXTERNAL temperature cells 1,2,3 respectively.
Cell 0 external temperature is mapped to the TH6 sensor as you mention.

TH6 is also appears to be affected by the heat generated from the balance board (when fitted) as its directly below that.

I was expecting that the all-in-one-monitoring-board (need to change that name:-)) would be fitted inside the same battery enclosure, so inside the thermal envelope/insulation.

Perhaps TH6 is of no real use?  Moving TH6 to internal cell0 could be a solution.

stuartpittaway, 2023-09-29:
I've made the changes to move TH6 to the internal cell0 measurement.  This is in commit 2b1dc07a2a45186b32c34ffd4d11b73e9e6e9a2e

and the compiled code will be in this GITHUB Action artifact https://github.com/stuartpittaway/diyBMSv4ESP32/actions/runs/6349861062

stuartpittaway, 2023-10-30:
Hi @delboy711 did you test the newest firmware ?

delboy711, 2023-11-05:
Sorry for the delay getting back Stuart.  I have been building a second board to use with my test battery. That is now complete and I have tested the new firmware and it works perfectly. Thanks

closed 2023-11-06 by stuartpittaway